wb2osz/direwolf #406: Digipeated messages contains NUL charachter

F4FXL, 2022-07-01:
_**Steps to reproduce**_
Send a message from RF to RF using a direwolf digipeater configured as digipeater only

_**Expected result**_
Message is digipeated unchanged and reaches its recipient

_**Actual Result**_
The message is corrupted. The { character marking the beginning of the message is replaced by a NUL charachter. Generally speaking it seems like message digipeating/igating is quite broken in Direwolf 1.7 see also #405

[image]

F4FXL, 2022-07-01:
It seems like the issue only occurs when igating is enabled... When igating is disbled it seems to be ok.... Not sure though, needs confirmation.

wb2osz, 2022-07-04:
Thank you for providing a clear complete description including an example to demonstrate the problem.
This has been fixed in the dev branch.

closed 2022-07-04 by wb2osz

wb2osz, 2023-01-28:
commit 30869c7afc5517eaf9f5a31bd21274215a4c5225
Date:   Mon Jul 4 02:11:55 2022 +0100

    Issues 405 & 406 and other improvements in message interpretation.

This is in the "dev" branch which will become release 1.7.

wb2osz, 2023-01-28:
The error was introduced in the dev branch (release 1.7 development) by commit e10814765539ef4db92c8d7095e8375e10421380 on February 25, 2022.

Those running release 1.6 or earlier not be impacted. This would affect only those who 
are using the "dev" branch from between the two dates mentioned.