wb2osz/direwolf #408: Direwolf v1.6 truncates Source SSID when IS to TX

joshbradfield, 2022-07-02:
When forwarding from the internet (IG) to radio (TX), any text following a hyphen "-" in the "source" call sign is dropped. 
e.g. `WHO-IS` becomes `WHO`

_Steps to reproduce:_
1. Setup direwolf as an RX/TX IGATE
2. From a message capable APRS device emit a beacon (so direwolf knows the device is in range)
3. From the same device emit a "WHO-IS" request/message

_Expected Result_
- Message ACK is returned to originator via RF, the source station should be "WHO-IS"
- WHO-IS response is returned to originator via RF, the source station should be "WHO-IS"

_Actual Result_
- Message ACK and Response is returned to originator via RF, but the "Source" station is incorrect (truncated) "WHO" instead of "WHO-IS"

### Example

WHO-IS request sent from ZL1JSH-9 via ZL1JSH-10,

```
[ig>tx] WHO-IS>APJIW4,TCPIP*,qAC,AE5PL-JF::ZL1JSH-9 :Charles Beadfield/New Zealand{583
mheard_save_is: WHO - update time, was 1656771162 seconds ago.
callsign  cnt chan hops    RF      IS    lat     long  msp
WHO         7   0   0      -      0:00     -       -    1
ZL1JSH-9   10   0   0     0:00     -    -36.79  174.76  0
ZL1NL-3     6   0   0     0:03     -    -36.84  174.77  0
ZL1JSH      2   0   0      -      0:16     -       -    1
Was message addressee ZL1JSH-9 heard in the past 60 minutes, with 1 or fewer digipeater hops?
Yes, ZL1JSH-9 last heard over radio 0 minutes ago, 0 digipeater hops.
Was message source WHO heard in the past 1 minutes, with 0 or fewer digipeater hops?
No, we have not heard WHO over the radio.
MSP for WHO set to 1
[0L] ZL1JSH-10>APDW16,WIDE1-1:}WHO>APJIW4,TCPIP,ZL1JSH-10*::ZL1JSH-9 :ack5
[0L] ZL1JSH-10>APDW16,WIDE1-1:}WHO>APJIW4,TCPIP,ZL1JSH-10*::ZL1JSH-9 :Charles Beadfield/New Zealand{583
```

wb2osz, 2022-07-04:
Thank you for reporting this problem.
The code was written, expecting only a numeric SSID.
This is an unusual, but perfectly valid case.
The message addressee, and therefore the sender of a reply, does not have the same restriction as the AX.25 addresses.

joshbradfield, 2022-07-04:
Yes, I should probably update this to make it clear that

*Issue only applicable for non-numeric SSIDs*

Otherwise, very much enjoying using direwolf :)