wb2osz/direwolf #415: UDP or TCP output

zeke800, 2022-07-15:
Hello,

Thank so much for the software! 

I have a very simple question - would it be possible to output decoded data to UDP or TCP?

Thank you!

dranch, 2022-07-15:
Direwolf issues are not for getting support, they are for submitting bug reports or feature requests.  Please join the direwolf@groups.io email list and submit your questions there and CLOSE this Github issue.  That said, it's not clear what you mean by "decoded data".  Are you talking about maybe a) the decoded APRS packets seen on STDOUT?  Are you talking about b) the payload of basic AX.25 packets?  If you're thinking of option a), you could pipe the output of Direwolf's STDOUT into a tool like socat which can send that data to any remote TCP, UDP, etc. receiver.

zeke800, 2022-07-15:
Thank you very much for responding! I'm sorry, I didn't know, as usually issues are the way to do this. 

I am talking about the decoded APRS packets. 

Thank you! Closing.

closed 2022-07-15 by zeke800