wb2osz/direwolf #418: Error when specifying rig 2027 in direwolf

adespain, 2022-08-28:
I am using a Kenwood TM-V71A with the PG-5h audio and serial cables connected to a Raspberry pi 3B 1.2  Trying to get PTT to work with Hamlib since the PTT /dev/ttyUSB0 RTS -DTR line in the direwolf.conf file didn't work.

I have this line in my direwolf.conf:

PTT RIG 2027 /dev/ttyUSB0 9600

When Direwolf starts up it shows this error in red:

Dire Wolf version 1.6
Includes optional support for:  hamlib cm108-ptt

Reading config file direwolf.conf
Audio device for both receive and transmit: plughw:3,0  (channel 0)
Channel 0: 1200 baud, AFSK 1200 & 2200 Hz, E+, 44100 sample rate / 3.
Hamlib determined CAT control serial port rate of 57600.
User configuration overriding hamlib CAT control speed to 9600.
Hamlib Rig open error -5: Communication timed out
kenwood_open: no response from rig
kenwood.c(872):kenwood_open return(-5)
rig.c(1031):rig_open return(-5)

When trying to transmit it says this:

Hamlib: Can't use rig_set_ptt for channel 0 PTT because rig_open failed.

I have tried using RIG 2026 as well with the same results.  I have been able to use the FLRIG gui and can click on the PTT button and it keys the radio just fine.  So I know the serial cable is working.  Any help would be appreciated, thanks!

dranch, 2022-08-29:
Hello adespain,

Github issues are not for getting support.  Please join the direwolf@groups.io email list, re-post your message there to get help, and close this issue here on Github.  When you do post there, please explain why you want to use CAT-based PTT control vs. say just using the RTS line from your USB to serial adapter?  Please also mention what you're using for a sound device as configured as plughw:3,0 as a Raspberry Pi itself doesn't include any microphone functionality itself.  One last thing, make sure whatever unix account you're running direwolf from is apart of the "dialout" unix group as mentioned in the Direwolf User Guide.

--David

adespain, 2022-08-29:
Will do, thank you!

closed 2022-08-29 by adespain