wb2osz/direwolf #421: Direwolf 1.5 adding PHG5100327 to wx pckets

jondkelley, 2022-09-11:
Issue noticed in Direwolf 1.5

I have a script that generates wxnow

```
# curl -s http://192.168.1.250:5000/wxnow.txt | tail -1
325/000g000t067r000p000P000h077b10118N5IPT Hi from Westwood in Kalamazoo, MI!*digi-gate*Direwolf on Rasp.Pi 1[2011]***see my weather station open source Python code @ https://github.com/jondkelley/eco_aprs_weather {ECOWITT}
```
I also have a beacon setup

```
PBEACON sendto=0 delay=0:30 every=17:17 power=30 height=25 gain=11 lat=42.303690 long=-85.628359 SYMBOL="weather station" COMMENTCMD="curl -s http://192.168.1.250:5000/wxnow.txt | tail -1"
```

For some reason my packets are showing up with a weird string `PHG5100327`
dw logs
```
[0L] WESTWD>APDW15:!4218.22N/08537.70W_PHG5100325/000g000t067r000p000P000h077b10120N5IPT Hi from Westwood in Kalamazoo, MI!*digi-gate*Direwolf on Rasp.Pi 1[2011]***see my weather station open source Python code @ https://github.com/jondkelley/eco_aprs_weather {ECOWITT}
[ig] WESTWD>APDW15:!4218.22N/08537.70W_PHG5100325/000g000t067r000p000P000h077b10120N5IPT Hi from Westwood in Kalamazoo, MI!*digi-gate*Direwolf on Rasp.Pi 1[2011]***see my weather station open source Python code @ https://github.com/jondkelley/eco_aprs_weather {ECOWITT}
```

findu 
```
WESTWD>APDW15,TCPIP*,qAC,T2CAEAST:!4218.22N/08537.70W_PHG5100327/000g000t067r000p000P000h068b00000N5IPT Hi from Westwood in Kalamazoo, MI!*digi-gate*Direwolf on Rasp.Pi 1[2011]***see my weather station open source Python code @ https://github.com/jondkelley/eco_aprs_weather {ECOWITT}
````

I think this string makes weather packet decode impossible on aprs websites.

Any idea what causes this?

jondkelley, 2022-09-11:
I now suspect it is 
```
power=30 height=25 gain=11
````

closed 2022-09-11 by jondkelley

dranch, 2022-09-11:
Please note that Direwolf v1.5 is very old.  Please rebuild using the "dev" branch here in Github as it's effectively the upcoming 1.7 version which has a lot of bug fixes, improvements, etc.

wb2osz, 2022-09-11:
Weather report with position can't have PHG.  Here is an easy way to see if a packet is valid, and what it means.

echo 'aprspacket'  | decode_aprs

example:

 echo 'WESTWD>APDW15:!4218.22N/08537.70W_PHG5100325/000g000t067r000p000P000h077b10120N5IPT Hi from Westwood in Kalamazoo, MI!*digi-gate*Direwolf on Rasp.Pi 1[2011]***see my weather station open source Python code @ https://github.com/jondkelley/eco_aprs_weather {ECOWITT}' | decode_aprs

WESTWD>APDW15:!4218.22N/08537.70W_PHG5100325/000g000t067r000p000P000h077b10120N5IPT Hi from Westwood in Kalamazoo, MI!*digi-gate*Direwolf on Rasp.Pi 1[2011]***see my weather station open source Python code @ https://github.com/jondkelley/eco_aprs_weather {ECOWITT}
Didn't find wind direction in form c999.
Didn't find wind speed in form s999.
Didn't find wind gust in form g999.
Didn't find temperature in form t999.
Weather Report, WEATHER Station (blue), DireWolf, WB2OSZ
N 42 18.2200, W 085 37.7000
, "PHG5100325/000g000t067r000p000P000h077b10120N5IPT Hi from Westwood in Kalamazoo, MI!*digi-gate*Direwolf on Rasp.Pi 1[2011]***see my weather station open source Python code @ https://github.com/jondkelley/eco_aprs_weather {ECOWITT}"

That was not recognized as a valid weather report.  Now remove the PHG5100..

 $ echo 'WESTWD>APDW15:!4218.22N/08537.70W_325/000g000t067r000p000P000h077b10120N5IPT Hi from Westwood in Kalamazoo, MI!*digi-gate*Direwolf on Rasp.Pi 1[2011]***see my weather station open source Python code @ https://github.com/jondkelley/eco_aprs_weather {ECOWITT}' | decode_aprs

WESTWD>APDW15:!4218.22N/08537.70W_325/000g000t067r000p000P000**_h077_**b10120N5IPT Hi from Westwood in Kalamazoo, MI!*digi-gate*Direwolf on Rasp.Pi 1[2011]***see my weather station open source Python code @ https://github.com/jondkelley/eco_aprs_weather {ECOWITT}
Weather Report, WEATHER Station (blue), DireWolf, WB2OSZ
N 42 18.2200, W 085 37.7000
wind 0.0 mph, direction 325, gust 0, temperature 67, rain 0.00 in last hour, rain 0.00 in last 24 hours, rain 0.00 since midnight, humidity 7, "7b10120N5IPT Hi from Westwood in Kalamazoo, MI!*digi-gate*Direwolf on Rasp.Pi 1[2011]***see my weather station open source Python code @ https://github.com/jondkelley/eco_aprs_weather {ECOWITT}"

The weather was MOSTLY decoded properly. 

Why was  "**_7b10120_**"  considered leftover?    If we look at the WXNOW.TXT format https://cumuluswiki.org/a/Wxnow.txt "h" should be followed by only two digits.   You have three and the barometric pressure was not decoded.

If we remove the extra '0', after the 'h', we get:

Humidity is now decoded properly.

echo 'WESTWD>APDW15:!4218.22N/08537.70W_325/000g000t067r000p000P000h77b10120N5IPT Hi from Westwood in Kalamazoo, MI!*digi-gate*Direwolf on Rasp.Pi 1[2011]***see my weather station open source Python code @ https://github.com/jondkelley/eco_aprs_weather {ECOWITT}' | decode_aprs

WESTWD>APDW15:!4218.22N/08537.70W_325/000g000t067r000p000P000h77b10120N5IPT Hi from Westwood in Kalamazoo, MI!*digi-gate*Direwolf on Rasp.Pi 1[2011]***see my weather station open source Python code @ https://github.com/jondkelley/eco_aprs_weather {ECOWITT}
Weather Report, WEATHER Station (blue), DireWolf, WB2OSZ
N 42 18.2200, W 085 37.7000
wind 0.0 mph, direction 325, gust 0, temperature 67, rain 0.00 in last hour, rain 0.00 in last 24 hours, rain 0.00 since midnight, humidity 77, **_barometer 29.89_**, "N5IPT Hi from Westwood in Kalamazoo, MI!*digi-gate*Direwolf on Rasp.Pi 1[2011]***see my weather station open source Python code @ https://github.com/jondkelley/eco_aprs_weather {ECOWITT}"

This is now correct.  you need to have only two digits for humidity.

wb2osz, 2022-09-11:
Here's your problem:

fields.append("h%03d" % (int(humidity_outdoor)))

Humidity is TWO digits.