wb2osz/direwolf #423: AGW protocol documentation

scd31, 2022-09-25:
I would like to write an application that uses AX.25. My understanding is that using AGW is the recommended way to do this rather than KISS. Am I correct in thinking this? Are there recommended documents I can look at to implement an AGW client?

wb2osz, 2022-09-25:
KISS is adequate for APRS which uses only UI frames.
If you want to access the connection oriented protocol implementation within direwolf, you need to use the AGW network interface.
A sample library is here:  https://github.com/wb2osz/direwolf/blob/dev/src/agwlib.c 
An example using it is here:  https://github.com/wb2osz/direwolf/blob/dev/src/tnctest.c  

73,
John WB2OSZ

closed 2022-09-25 by wb2osz

dranch, 2022-09-26:
Just as a follow up, Github "issues" are not intended for getting tech support.  If you need additional help, please join the direwolf@groups.io email list where you will find lots of helpful people for everything Direwolf related.