wb2osz/direwolf #425: Direwolf v1.6 - IGate->RF packets malformed?

Tyler-2, 2022-09-30:
I noticed that when I send a message from my phone (APRSDroid) via APRS-IS, when my Direwolf digi picks it up and then tries to TX to RF, it seems to mangle the "header":

```
[ig>tx] K1MLN-7>APDR16,TCPIP*,qAC,T2MCI::K1MLN-1  :Test{4
[0L] K1MLN-2>APDW16,WIDE1-2:}K1MLN-7>APDR16,TCPIP,K1MLN-2*::K1MLN-1  :Test{4
```

I think this should look more like:
```
[ig>tx] K1MLN-7>APDR16,TCPIP*,qAC,T2MCI::K1MLN-1  :Test{4
[0L] K1MLN-7>APDR16,TCPIP,K1MLN-2*,WIDE1-2::K1MLN-1  :Test{4
```

Shouldn't it?

The current case doesn't look like it would be routable, and indeed, I don't see an indication that my other APRS digi picks it up.

wb2osz, 2022-09-30:
Thanks for writing.

"WIDE1-2" is highly unusual.  Some digipeaters might discard it depending on the configuration.
You probably want to change the IGate configuration to use "WIDE1-1,WIDE2-1".
Otherwise it looks normal.

An IGate station wraps the original packet in a third party header.
That way it has both the RF station transmitting this new packet and it retains the original sending station.
When properly functioning APRS implementations see the "}" data type indicator, they process everything after it to obtain a modified version of the original packet.

You might find these references interesting:

APRS-IS (Automatic Packet Reporting System-Internet Service)
http://www.aprs-is.net/Default.aspx

APRS iGate properties
http://wiki.ham.fi/APRS_iGate_properties
(now gone but you can find a copy here:)
https://web.archive.org/web/20120503201832/http://wiki.ham.fi/APRS_iGate_properties

Notes to iGate developers
https://github.com/hessu/aprsc/blob/master/doc/IGATE-HINTS.md#igates-dropping-duplicate-packets-unnecessarily

Tyler-2, 2022-10-02:
Thank you, looks like this format wasn't a mistake, it's just something I'm unfamiliar with. I'll read up on the docs. Two things I've already learned:

1. Just because you're sending stuff to APRS-IS doesn't mean you can crank up the beacon rates.
2. The main reason I even noticed this was that my nearby digi wasn't digipeating my igate->RF packets.... But it is probably because it had a filter configured to not do so. I'll correct that next time I have access to it.

Thanks for the careful and informative reply, it was probably more than I deserved!

closed 2022-10-02 by Tyler-2