wb2osz/direwolf #437: Getting error: Timeout waiting for input from audio device 0

MatCat, 2022-12-30:
I am using SDR Console piping audio out on  a virtual audio cable, and have direwolf listening to that virtual audio cable output, it properly decodes my own APRS packets, however after 60 seconds direwolf quits with the error:

Timeout waiting for input from audio device 0.
Terminating after audio input failure.

MatCat, 2022-12-30:
After some experimenting it is specifically at the timing of whatever pbeacon delay is that this error happens and it quits, any idea why?

dranch, 2022-12-30:
Direwolf issues are not the right place to get enduser support.  Please subscribe to direwolf@groups.io and re-post your problem there and I'm sure many helpful people will get you going.  In that post, please include all the text you see when Direwolf first starts.  Please close this Github issue.

MatCat, 2022-12-30:
Not a fan of email lists, is there a forum or something I can ask on?

dranch, 2022-12-30:
groups.io has a good web interface and has an excellent search ability as well.  For example:

   https://groups.io/g/direwolf/search?p=created%2C0%2Cvirtual+audio+cable%252C%2C20%2C2%2C0%2C7348&q=virtual+audio+cable    

Regardless, the groups.io system is all that exists for *official* Direwolf support.

wb2osz, 2022-12-30:
As David said, this place is for tracking defects and enhancement requests.  Questions like this should go to the discussion forum. 
Your description is rather vague.  More details would be helpful to figure out what is happening.
I'm not familiar with "SDR Console" (assuming to be https://www.sdr-radio.com/Console ) so posing the question to a forum with over 1500 users would be a better way to get help.  
It looks like this application is only for Windows.  What sort of virtual cable are you using?  What do you have in the direwolf configuration file to attach to this virtual audio cable?  It's very interesting that it would work for 60 seconds then quit. Try running Task Manager to see how much CPU time is being used.  Tell us more about the PBEACON.  What happens if you disable that?
73, John WB2OSZ

closed 2022-12-30 by wb2osz