wb2osz/direwolf #444: Test Tones mismatch with text on screen

F4FXL, 2023-01-28:
Steps to reproduce

- start direwolf with option -xm, 2200Hz Tone is sent whereas on screen text says 1200Hz
- start direwolf with option -xs, 1200Hz tone is sent whereas screen text says 2200Hz

dranch, 2023-01-28:
Thank you for creating this git issue.  I can confirm your finding is correct and this was first reported by another user (Marcel Eschenmoser) back on 11/18/22 on the Direwolf email list.  I did verify this issue and I reported this to WB2OSZ about this back on 11/18/2022 via email.  I don't think this has any impact on performance and the issue is only related to the calibrate function.

wb2osz, 2023-01-28:
Thanks for pointing this out.
Fixed in:

commit 031c937cdb123d2501180b6dbf1cc6cff9be3e2f
Date:   Sat Jan 28 16:36:51 2023 +0000

closed 2023-01-28 by wb2osz