wb2osz/direwolf #447: Direwolf logo/icon

TheSkorm, 2023-02-22:
I find the direwolf logo a little weird/creepy/scary. I was wondering if you'd be interested in replacing with something more cute a fluffy if I was to commission an artist?

philpem, 2023-02-22:
I didn't even realise this project had a logo. I can't see one linked from the README.md or the user guide.

Do you mean the photo of the direwolf skeleton on the first page of the user guide PDF?

TheSkorm, 2023-02-22:
The icon I'm referring to is https://github.com/wb2osz/direwolf/blob/master/cmake/cpack/direwolf_icon.png and the ico file
 which shows up in window managers and application selectors in some linux distros.

[image]

kd8drx, 2023-03-08:
I’d happily contribute funds to the effort! The icon is workable for sure, but definitely has a “made in MSPaint as a placeholder” vibe to it.

Besides, every OSS project needs an awesome mascot.

wb2osz, 2023-03-08:
The icon is found in:
* a linux desktop icon to run the app.
* Windows executable seen in some contexts (this might have been broken when going to cmake).
* The powerpoint presentations,  https://github.com/wb2osz/direwolf-presentation

It is a possibility as long it is original and not just copied from the Internet. ( https://duckduckgo.com/?q=direwolf&t=ffab&atb=v345-1&iar=images&iax=images&ia=images )   What did you have in mind?

mikeym88, 2023-03-28:
What about this one: https://iconduck.com/icons/25068/direwolf?

wb2osz, 2023-03-31:
Thanks for the suggestion, but ...

That is part of a generic collection and could be used many other places.  We don't know where it came from originally and whether there could be a copyright violation.  I want to have something original and unique so there is no confusion or other potential issues.

mikeym88, 2023-04-01:
Just to note: it uses the Creative Commons license so you don't need to worry about copyright.

TheSkorm, 2023-04-01:
I'm working on getting some art work commissioned (no obligations if it ends up not suitable) but it'll take some time. This should satisfy the constraints of being unique and copyright would remain with wb2osz.