wb2osz/direwolf #450: could not open device

wf5w, 2023-02-23:
direwolf version: latest dev branch cloned as of Feb 23rd, 2023
 
OS: Linux Mint 21
device: digirig 1.9 [ logic level type] 
rigs: both FT-2980 and FT-8800 exhibit this behavior

$ groups
adm tty mail dialout fax cdrom sudo audio dip plugdev lpadmin ssh sambashare vboxusers

$ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: PCH [HDA Intel PCH], device 0: CX20756 Analog [CX20756 Analog]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 3: Device [USB PnP Sound Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

$ dmesg
[   48.280432] usb 2-3.2: cp210x converter now attached to ttyUSB0

direwolf.conf:

ADEVICE plughw:3,0
CHANNEL 0
MODEM 1200
PTT /dev/ttyUSB0 RTS

$ dirwolf -p

Could not open audio device plughw:3,0 for output
Device or resource busy
Pointless to continue without audio device.

Not sure why I should get this error.

Any ideas on what to do?

Jerry

dranch, 2023-02-23:
Hello Jerry,
The use of Git "issues" is really not a place to get support.  I recommend you join the direwolf@groups.io email list and re-post this last question there to help from lots of people including those who are using the digirig device.  In that email post, please also include the output of "aplay -l" as well.  It might also be worth trying to run direwolf as root just to see if the behavior changes as well.

wb2osz, 2023-02-23:
Could not open audio device plughw:3,0 for output
Device or resource busy

So, it looks like some other application is using the audio device.
Apparently you are running this with systemd.

How about simplifying the situation and just running it from the command line?
That might provide some clues.

And, yes, David is correct.  This is a place for reporting software defects or requesting enhancements, no general troubleshooting.  You might find the issue was already addressed in the forum.  Or the solution you find will help others.

wf5w, 2023-02-24:
ok. I will go to the groups to do this.
I suspect it has something to do with pulse audio getting in the way.

wb2osz, 2023-05-19:
When the "Device or resource busy" message appears, this additional explanation is added:

This means that some other application is using that device.
The solution is to identify that other application and stop it.

closed 2023-05-19 by wb2osz