wb2osz/direwolf #452: Suppress the volume too high when audio source is a stream

wb2osz, 2023-02-28:
Typically, I use RTL_FM and feed the stdout to Direwolf when working with an RTL dongle (typically just for testing).
However, I decide to give gqrx a try and the UDP output to Direwolf.  Everything works great, but I get a 'Volume too high' error with the UDP data.
I looked though the gqrx info and can't find any volume control.

Anyone experience this and found a solution?
I haven't asked ion any gqrx circles - yet.
-------
Rob KB8RCO

guitarpicva, 2023-04-27:
Been messing with this myself today.  See latest Issue #468.

guitarpicva, 2023-04-27:
Confirmed that GQRX output make Direwolf report a 200 value for audio level, which only goes down when it receives something via the UDP socket.  I can't find a way to lower GQRX output for streaming either.  SDR++ does have such a control, but then it doesn't have any obvious way to ensure 16 LE signed.