wb2osz/direwolf #453: meshtastic

srem1, 2023-03-08:
Is possible integrate with meshtastic no licence network?
https://meshtastic.org/
or https://reticulum.network/index_pl.html

dranch, 2023-03-08:
I don't think this would  be possible using the Meshtastic system as it's based on LORA technology which is patented. 

https://github.com/wb2osz/direwolf/issues/453
--
LoRa is proprietary and patented by Semtech Corporation. LoRaWAN is built on top of LoRa and is standardized by LoRa Alliance. 
--

wb2osz, 2023-03-09:
David -  I don't understand your comment.
It would be possible to build a gateway between different communication systems conveying positions and messages.
Or the APRS protocol could be divided into layers and the application layer could be combined with a different data link layer.

https://www.youtube.com/watch?v=43FD9CvXjh0

dranch, 2023-03-09:
Maybe we're looking at this from different perspectives and it wasn't clear what you are looking for.  

My comment was regarding having Direwolf operate as a standalone meshtasic node which would include the LORA modem, the upper communication layer, etc.  Since the LORA modem is patented and only available via silicon chips today, that would bring in all kinds of legal issues.  Generally speaking, Direwolf is fundamentally a modem first and it then add various protocol stacks on the side (AX.25, APRS, APRS-IS, AGW, KISS, etc).  

Can you more completely describe what are you looking for?

fab1an2, 2024-06-21:
reticulum have many interfaces https://reticulum.network/manual/interfaces.html
pipe, or kiss etc.
Question is how configuring direwolf for communication.
In my opinion tutorial how configuring in PMR , 433MHz will be great.