wb2osz/direwolf #468: Is the EAS modem encode/decode and fully functional in "dev" branch?

guitarpicva, 2023-04-27:
I have been streaming UDP data from GQRX or SDR++ into direwolf to try and decode the EAS Codec which is sent by the minimodem application from Kamal Mustaffa.  

Has EAS been fully tested for rx and tx as currently implemented in the dev branch?  So far, direwolf gives no indication that it hears the data coming in.  It DOES show that it is set up for the Codec upon startup.  I have verified with wireshark that the samples are on UDP:localhost:7355.

Direwolf seems to bind directly to port 7355 also which seems odd to me. (Maybe I am just used to using multicast).

mitch@ondev:~/src$ netstat -anu
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State      
udp        0      0 0.0.0.0:7355            0.0.0.0:*                        

mitch@ondev:~$ direwolf -t 0 -b 16
Dire Wolf DEVELOPMENT version 1.7 F (Apr 27 2023)
Includes optional support for:  hamlib cm108-ptt

Reading config file direwolf.conf
Audio input device for receive: UDP:7355  (channel 0)
Audio out device for transmit: default  (channel 0)
Channel 0: FIX_BITS option has been turned off for EAS.
Channel 0: 521 baud, AFSK 2083 & 1563 Hz, A+, 48000 sample rate / 3.
Note: PTT not configured for channel 0. (Ignore this if using VOX.)
Disabled AGW network client port.
Ready to accept KISS TCP client application 0 on port 8001 ...

wb2osz, 2023-04-27:
direwolf decodes National Weather Service EAS SAME signals fine,  A couple other people have mentioned success.
It does not transmit EAS SAME because there is no real need and the capability could be abused.
I've never tried using minimodem so I have no idea whether it is generating valid EAS SAME signals.

Did you read the documentation here?
https://github.com/wb2osz/eas2aprs

guitarpicva, 2023-04-27:
Actually, there is a need.  We are testing SAME as a national alerting format and use of the same codec only makes sense.

The "magic" is not the waveform since it's just an odd RTTY.  The format and timing of SAME messages used by NWS is how that whole system works.  It's all fully documented in public.

wb2osz, 2023-04-27:
Yes, I know that the format is publicly documented. That is how I was able to write a decoder.

Tell me more about your need to transmit it.

It shouldn't be hard.  But, what would the benefit be to the community?

guitarpicva, 2023-04-27:
Sent a private email John.

closed 2023-04-27 by guitarpicva

k8tek, 2024-06-03:
Direwolf is not parsing EAS data correctly.  It is not inserting the closing curly bracket.

{DEZCZC-EAS-RWT-012057-012081-012101-012103-012115+0030-2780415-WTSP/TV-

wb2osz, 2024-06-03:
Could you provide more detail?

KR8MER, 2024-06-04:
> Could you provide more detail?

Yes, when I receive EAS alerts over the local NOAA WR channel, it decodes the header just fine, but this is what it decodes....

{DEZCZC-EAS-RWT-012057-012081-012101-012103-012115+0030-2780415-WTSP/TV-

Notice the closing bracket ( } )is missing.  EAS2APRS expects the closing bracket, which is absent.  This is causing EAS2APRS to not decode it and pass it on to DSAME. It is decoding it the same way for all 3 decodes.

Running Direwolf 1.8

KR8MER, 2024-06-05:
Here is a decoded packet for EAS

2024-06-05 11:01:14 EDT: [EAS](https://aprs.fi/?c=raw&limit=&call=EAS)>APDW18,qAR,[EAS](https://aprs.fi/?c=raw&limit=&call=EAS):{DEZCZC-WXR-RWT-039065-039175-039011-039161-039063-039039-039149-039091-039107-039003-039125-039137+0600-1571500-KILN/NWS-

Notice Direwolf is not closing the message with a **}**

wb2osz, 2024-06-05:
This uses the user-defined APRS packet format.

Data Type Indicator of { means user defined.
D is for direwolf application
E is for EAS data.
The EAS sentence is:

ZCZC-EAS-RWT-012057-012081-012101-012103-012115+0030-2780415-WTSP/TV-

73,
John WB2OSZ