wb2osz/direwolf #469: Can't build on Windows

iontodirel, 2023-05-09:
I am trying to build Direwolf on Windows with cygwin. Compile succeeds, but link fails:

[ 40%] Linking C executable direwolf.exe
/usr/lib/gcc/x86_64-w64-mingw32/11/../../../../x86_64-w64-mingw32/bin/ld: CMakeFiles/direwolf.dir/ax25_link.c.o:ax25_link.c:(.bss+0x0): multiple definition of `re_syntax_options'; CMakeFiles/direwolf.dir/direwolf.c.o:direwolf.c:(.bss+0x0): first defined here
/usr/lib/gcc/x86_64-w64-mingw32/11/../../../../x86_64-w64-mingw32/bin/ld: CMakeFiles/direwolf.dir/ax25_pad.c.o:ax25_pad.c:(.bss+0x0): multiple definition of `re_syntax_options'; CMakeFiles/direwolf.dir/direwolf.c.o:direwolf.c:(.bss+0x0): first defined here
/usr/lib/gcc/x86_64-w64-mingw32/11/../../../../x86_64-w64-mingw32/bin/ld: CMakeFiles/direwolf.dir/beacon.c.o:beacon.c:(.bss+0x0): multiple definition of `re_syntax_options'; CMakeFiles/direwolf.dir/direwolf.c.o:direwolf.c:(.bss+0x0): first defined here
/usr/lib/gcc/x86_64-w64-mingw32/11/../../../../x86_64-w64-mingw32/bin/ld: CMakeFiles/direwolf.dir/config.c.o:config.c:(.bss+0x0): multiple definition of `re_syntax_options'; CMakeFiles/direwolf.dir/direwolf.c.o:direwolf.c:(.bss+0x0): first defined here
/usr/lib/gcc/x86_64-w64-mingw32/11/../../../../x86_64-w64-mingw32/bin/ld: CMakeFiles/direwolf.dir/decode_aprs.c.o:decode_aprs.c:(.bss+0x0): multiple definition of `re_syntax_options'; CMakeFiles/direwolf.dir/direwolf.c.o:direwolf.c:(.bss+0x0): first defined here
/usr/lib/gcc/x86_64-w64-mingw32/11/../../../../x86_64-w64-mingw32/bin/ld: CMakeFiles/direwolf.dir/dedupe.c.o:dedupe.c:(.bss+0x0): multiple definition of `re_syntax_options'; CMakeFiles/direwolf.dir/direwolf.c.o:direwolf.c:(.bss+0x0): first defined here
/usr/lib/gcc/x86_64-w64-mingw32/11/../../../../x86_64-w64-mingw32/bin/ld: CMakeFiles/direwolf.dir/digipeater.c.o:digipeater.c:(.bss+0x0): multiple definition of `re_syntax_options'; CMakeFiles/direwolf.dir/direwolf.c.o:direwolf.c:(.bss+0x0): first defined here
/usr/lib/gcc/x86_64-w64-mingw32/11/../../../../x86_64-w64-mingw32/bin/ld: CMakeFiles/direwolf.dir/cdigipeater.c.o:cdigipeater.c:(.bss+0x0): multiple definition of `re_syntax_options'; CMakeFiles/direwolf.dir/direwolf.c.o:direwolf.c:(.bss+0x0): first defined here
/usr/lib/gcc/x86_64-w64-mingw32/11/../../../../x86_64-w64-mingw32/bin/ld: CMakeFiles/direwolf.dir/dwgps.c.o:dwgps.c:(.bss+0x0): multiple definition of `re_syntax_options'; CMakeFiles/direwolf.dir/direwolf.c.o:direwolf.c:(.bss+0x0): first defined here
/usr/lib/gcc/x86_64-w64-mingw32/11/../../../../x86_64-w64-mingw32/bin/ld: CMakeFiles/direwolf.dir/igate.c.o:igate.c:(.bss+0x0): multiple definition of `re_syntax_options'; CMakeFiles/direwolf.dir/direwolf.c.o:direwolf.c:(.bss+0x0): first defined here
/usr/lib/gcc/x86_64-w64-mingw32/11/../../../../x86_64-w64-mingw32/bin/ld: CMakeFiles/direwolf.dir/kiss_frame.c.o:kiss_frame.c:(.bss+0x0): multiple definition of `re_syntax_options'; CMakeFiles/direwolf.dir/direwolf.c.o:direwolf.c:(.bss+0x0): first defined here
/usr/lib/gcc/x86_64-w64-mingw32/11/../../../../x86_64-w64-mingw32/bin/ld: CMakeFiles/direwolf.dir/kiss.c.o:kiss.c:(.bss+0x0): multiple definition of `re_syntax_options'; CMakeFiles/direwolf.dir/direwolf.c.o:direwolf.c:(.bss+0x0): first defined here
/usr/lib/gcc/x86_64-w64-mingw32/11/../../../../x86_64-w64-mingw32/bin/ld: CMakeFiles/direwolf.dir/kissserial.c.o:kissserial.c:(.bss+0x0): multiple definition of `re_syntax_options'; CMakeFiles/direwolf.dir/direwolf.c.o:direwolf.c:(.bss+0x0): first defined here
/usr/lib/gcc/x86_64-w64-mingw32/11/../../../../x86_64-w64-mingw32/bin/ld: CMakeFiles/direwolf.dir/kissnet.c.o:kissnet.c:(.bss+0x0): multiple definition of `re_syntax_options'; CMakeFiles/direwolf.dir/direwolf.c.o:direwolf.c:(.bss+0x0): first defined here
/usr/lib/gcc/x86_64-w64-mingw32/11/../../../../x86_64-w64-mingw32/bin/ld: CMakeFiles/direwolf.dir/waypoint.c.o:waypoint.c:(.bss+0x0): multiple definition of `re_syntax_options'; CMakeFiles/direwolf.dir/direwolf.c.o:direwolf.c:(.bss+0x0): first defined here
/usr/lib/gcc/x86_64-w64-mingw32/11/../../../../x86_64-w64-mingw32/bin/ld: CMakeFiles/direwolf.dir/pfilter.c.o:pfilter.c:(.bss+0x0): multiple definition of `re_syntax_options'; CMakeFiles/direwolf.dir/direwolf.c.o:direwolf.c:(.bss+0x0): first defined here
/usr/lib/gcc/x86_64-w64-mingw32/11/../../../../x86_64-w64-mingw32/bin/ld: CMakeFiles/direwolf.dir/recv.c.o:recv.c:(.bss+0x0): multiple definition of `re_syntax_options'; CMakeFiles/direwolf.dir/direwolf.c.o:direwolf.c:(.bss+0x0): first defined here
/usr/lib/gcc/x86_64-w64-mingw32/11/../../../../x86_64-w64-mingw32/bin/ld: CMakeFiles/direwolf.dir/server.c.o:server.c:(.bss+0x0): multiple definition of `re_syntax_options'; CMakeFiles/direwolf.dir/direwolf.c.o:direwolf.c:(.bss+0x0): first defined here
/usr/lib/gcc/x86_64-w64-mingw32/11/../../../../x86_64-w64-mingw32/bin/ld: CMakeFiles/direwolf.dir/tt_user.c.o:tt_user.c:(.bss+0x0): multiple definition of `re_syntax_options'; CMakeFiles/direwolf.dir/direwolf.c.o:direwolf.c:(.bss+0x0): first defined here
/usr/lib/gcc/x86_64-w64-mingw32/11/../../../../x86_64-w64-mingw32/bin/ld: CMakeFiles/direwolf.dir/xmit.c.o:xmit.c:(.bss+0x0): multiple definition of `re_syntax_options'; CMakeFiles/direwolf.dir/direwolf.c.o:direwolf.c:(.bss+0x0): first defined here
/usr/lib/gcc/x86_64-w64-mingw32/11/../../../../x86_64-w64-mingw32/bin/ld: CMakeFiles/direwolf.dir/dwgpsnmea.c.o:dwgpsnmea.c:(.bss+0x0): multiple definition of `re_syntax_options'; CMakeFiles/direwolf.dir/direwolf.c.o:direwolf.c:(.bss+0x0): first defined here
/usr/lib/gcc/x86_64-w64-mingw32/11/../../../../x86_64-w64-mingw32/bin/ld: ../external/regex/libregex.a(regex.c.o):regex.c:(.bss+0x0): multiple definition of `re_syntax_options'; CMakeFiles/direwolf.dir/direwolf.c.o:direwolf.c:(.bss+0x0): first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/direwolf.dir/build.make:1076: src/direwolf.exe] Error 1
make[1]: *** [CMakeFiles/Makefile2:528: src/CMakeFiles/direwolf.dir/all] Error 2
make: *** [Makefile:156: all] Error 2

dranch, 2023-05-09:
You haven't provided enough information to confirm/deny this is a bug:

- Windows details (Windows version and if it's 32 or 64bit)
- What version of Cygwin mingw?  32 or 64bit?
- What version of direwolf are you trying to compile?  
- Did you install all the dependencies and environment variables for Windows compiling per https://github.com/wb2osz/direwolf/blob/dev/doc/User-Guide.pdf

dranch, 2023-05-10:
Are you trying to compile Direwolf 1.6 (master GIT branch) or Direwolf 1.7G (DEV Git branch)?

wb2osz, 2023-05-10:
This clearly has something to do with the regular expression (regex) library.
On Linux, direwolf uses the version of regex supplied by the operating system.
A copy of regex is included for use on Windows where it does not exist.
The error message mentions duplicate definition of something from the regex library
I have Windows 10 here, and the versions of cygwin and mingw are not up to date.

This seems to be the culprit:   REGEX_VARIABLE_IMPEXP reg_syntax_t re_syntax_options;
Maybe something about multiple definitions of the same external variable has changed.

wb2osz, 2023-05-10:
Try this:

Edit   external/regex/regex.h

Change from:  REGEX_VARIABLE_IMPEXP reg_syntax_t re_syntax_options;

Change to:  extern reg_syntax_t re_syntax_options;

Let me know how it goes.

73,
john WB2OSZ

iontodirel, 2023-05-11:
I will try that tomorrow, thank you. And that seems like the right fix.

wb2osz, 2023-05-19:
Any results?

iontodirel, 2023-05-22:
Hi @wb2osz, that worked! And tested that Direwolf works

wb2osz, 2023-05-22:
I'm glad to hear that you were able to build it OK.

The cmake scripts were provided by someone else and they have extra stuff, from other projects, in there that we really don't need.
I've only tried using gnu and clang compilers to make portability easier.
I have no idea how much effort would be needed to make direwolf build with MSVC.
It probably wouldn't be too horrible because it is using the windows header files and C runtime API.
Compiler options would definitely need to be changed.

wb2osz, 2023-10-20:
Issue resolved.  No discussion for many months.

closed 2023-10-20 by wb2osz