wb2osz/direwolf #471: Audio device names for Windows

iontodirel, 2023-05-22:
On Linux, Direwolf works very nicely by taking card and stream names in the ALSA format like **plughw:0,1**. This name for an audio device is predictable and unambiguous while the system is running, so if I have 5 sound cards, it's not a problem to set the right one. In Windows however, the names are just numbers, if I have 5 sound cards in Windows, I don't know how to predictably use the correct one. Could we change this to take string names instead that we could set in the direwolf.config?

wb2osz, 2023-05-22:
Actually, the ALSA "card" numbers are not predictable.  If you add another audio device, or remove one, and then reboot, previously existing audio devices can have different "card" numbers.  It is possible to assign a name, based on characteristics such as vendor, product, and serial number.  For multiple identical devices, names can be assigned by physical USB port.  See https://github.com/wb2osz/direwolf-doc/blob/main/Radio-Interface-Guide.pdf   section 7.6.

As you pointed out, using a number on Windows is not the best approach because the numbers change.  You can use a substring of the audio device name.  For example, if you add a single USB audio adapter, simply use "ADEVICE USB".  This is mentioned in the User Guide, in the section named "Audio Device Selection - Windows."   Radio-Interface-Guide.pdf section 2.2 also mentions this.  If you have multiple identical USB audio devices, Windows gives them unique names. 

Suggestions are always welcome for improving the documentation.  There is a lot.  sometimes it's not easy to find the right section.

iontodirel, 2023-05-22:
@wb2osz FYI, I created this utility https://github.com/iontodirel/find_devices, it’s only for Linux, but it reliably finds ALSA names based on names, description, serial numbers or hardware paths, when you have more than one sound card. It can also find reliably one of many Digirigs and their individual sound card / serial ports.

@wb2osz would you support contributions to improve the Windows device referencing story?