wb2osz/direwolf #472: Should AGWPE mirror Direwolf's RX/TX?

Tyler-2, 2023-05-25:
My Direwolf is configured as an RF->Igate and Igate->RF digipeater. Xastir connects via AGWPE.

If I send an APRS message from my phone (K1MLN-7) over APRS-IS,to my Xastir client (K1MLN), Direwolf gets it over the IS interface:

`May 25 12:54:05 direwolf direwolf[6143]: [ig>tx] K1MLN-7>APDR16,TCPIP*,qAC,T2SJC::K1MLN    :Ping4{30`

And then decides to gate it to radio, as expected:

```
May 25 12:54:05 direwolf direwolf[6143]: Was message addressee K1MLN heard in the past 90 minutes, with 2 or fewer digipeater hops?
May 25 12:54:05 direwolf direwolf[6143]: Yes, K1MLN last heard over radio 1 minutes ago, 2 digipeater hops.
May 25 12:54:05 direwolf direwolf[6143]: Was message source K1MLN-7 heard in the past 1 minutes, with 0 or fewer digipeater hops?
May 25 12:54:05 direwolf direwolf[6143]: No, we have not heard K1MLN-7 over the radio.
May 25 12:54:05 direwolf direwolf[6143]: [0L] K1MLN-1>APDW16,WIDE1-1,WIDE2-1:}K1MLN-7>APDR16,TCPIP,K1MLN-1*::K1MLN    :Ping4{30
```

However, my AGWPE connected Xastir client, `K1MLN`, does not see this message. It doesn't appear in `Incoming Data`. It _does_ eventually see it if Direwolf happens to hear another nearby Digipeater digipeat the message:

```
May 25 12:55:36 direwolf direwolf[6143]: [0L] K1MLN-1>APDW16,WIDE1-1,WIDE2-1:}K1MLN-7>APDR16,TCPIP,K1MLN-1*::K1MLN    :Ping4{30
May 25 12:55:39 direwolf direwolf[6143]: Digipeater WIDE1 (probably K2VIZ-8) audio level = 1(0/0)   [NONE]   _||||:___
May 25 12:55:39 direwolf direwolf[6143]: [0.3] K1MLN-1>APDW16,K2VIZ-8,WIDE1*,WIDE2-1:}K1MLN-7>APDR16,TCPIP,K1MLN-1*::K1MLN    :Ping4{30
May 25 12:55:39 direwolf direwolf[6143]: Third Party Header, Ambulance, DireWolf, WB2OSZ
May 25 12:55:39 direwolf direwolf[6143]: [0L] K1MLN>APX218,WIDE1-1,WIDE2-2::K1MLN-7  :ack30
```

So my AGWPE client doesn't see the packets that come in via IS... It also doesn't see packets that go out to RF or IS.

- Is it accurate to say that AGWPE clients will only hear RF packets received by Direwolf?
- Is it possible to ensure that AGWPE clients see RF packets _sent_ by Direwolf?
- Is it possible to ensure that AGWPE clients see IS packets sent and received by Direwolf?

wb2osz, 2023-05-26:
(1) With the default configuration, client applications only hear packets coming from RF.
(2) It would be possible if you operate in full duplex and have really bad audio crosstalk so you decode audio being sent to the transmitter.  But full duplex would be a bad idea because it would transmit immediately rather than waiting for a clear channel.
(3) There is a new configuration item, ICHANNEL, which maps a client app channel to APRS-IS.  This allows a client app to talk directly to APRS-IS.  It would not see packets processed by the RF>IS IGate.

You should find this helpful:   https://github.com/wb2osz/direwolf-doc/blob/main/Internal-Packet-Routing.pdf

Tyler-2, 2023-05-26:
Excellent.

I think you've answered all of my questions and basically:

(1) Yes
(2) No
(3) Soon - or today if you use `dev` branch (1.7)

It sounds like there would still not be a way for a connected client to see RF beacons sent by Direwolf, correct? And yeah, the crosstalk solution sounds bad. Ha

That packet routing doc is definitely useful.

mfncooper, 2024-07-12:
@Tyler-2 the AGWPE protocol does allow for transmitted frames to be provided to the client. The client first needs to explicitly enable monitoring, and then needs to listen for special "own" frames to receive transmitted frames in addition to received frames. Direwolf supports this, so it would be up to Xastir to take advantage of it.