wb2osz/direwolf #475: "Bad Address" errors

dancrossnyc, 2023-06-13:
I have two instances of direwolf communicating with two radios on a single machine and recently, transmissions has started failing with the following error:

```
Jun 13 13:00:01 vhf1.kz2x.ampr.org direwolf[424]: Audio output start error.
Jun 13 13:00:01 vhf1.kz2x.ampr.org direwolf[424]: Bad address
Jun 13 13:00:01 vhf1.kz2x.ampr.org direwolf[424]: Error preparing after bad state: Bad address
Jun 13 13:00:01 vhf1.kz2x.ampr.org direwolf[424]: Error preparing after bad state: Bad address
Jun 13 13:00:01 vhf1.kz2x.ampr.org direwolf[424]: Error preparing after bad state: Bad address
Jun 13 13:00:01 vhf1.kz2x.ampr.org direwolf[424]: Error preparing after bad state: Bad address
Jun 13 13:00:01 vhf1.kz2x.ampr.org direwolf[424]: Error preparing after bad state: Bad address
Jun 13 13:00:01 vhf1.kz2x.ampr.org direwolf[424]: Error preparing after bad state: Bad address
Jun 13 13:00:01 vhf1.kz2x.ampr.org direwolf[424]: Error preparing after bad state: Bad address
Jun 13 13:00:01 vhf1.kz2x.ampr.org direwolf[424]: Error preparing after bad state: Bad address
Jun 13 13:00:01 vhf1.kz2x.ampr.org direwolf[424]: Error preparing after bad state: Bad address
Jun 13 13:00:01 vhf1.kz2x.ampr.org direwolf[424]: Error preparing after bad state: Bad address
Jun 13 13:00:01 vhf1.kz2x.ampr.org direwolf[424]: Audio write error retry count exceeded.
```

Relevant platform details:
* Running on an x86_64 machine (Udoo BOLT v8 with quad-core AMD Ryzen CPUs, 16GiB RAM, 1TB NVMe SSD)
* Arch Linux (uname -a: `Linux vhf1.kz2x.ampr.org 6.3.7-arch1-1 #1 SMP PREEMPT_DYNAMIC Sat, 10 Jun 2023 00:35:35 +0000 x86_64 GNU/Linux`)
* Direwolf built from source on the "dev" branch (commit f9cf42b2918f4c0bf245dff03945b3e1407f7f28 plus a small local patch just to change hard-coded pathnames; there are no behavioral differences).
* Links against hamlib, also built from source (commit 

The receive path is fine on both instances.

Both instances of direwolf talk to Tigertronics Signalink USB external sound cards via ALSA; there is no pulseaudio installed on this machine.

Any advice is appreciated! I greatly prefer direwolf to a hardware TNC.

dranch, 2023-06-13:
Hello Dan,
This Github "issues" area is not intended for user support.  Please close this Github issue, join the Direwolf groups.io email list, and re-post your issue there.  That said, your above output including the line "Audio write error retry count exceeded" gives a major clue.  It looks like the sound devices are having issues due to probably RFI getting into the devices and/or cables.  If you're running two systems locally over RF, you must run at very low power, separate the devices  by some decent amount, etc.  It would also be advisable to add the correct RFI-suppressing ferrite chokes to all cables going in/out of the computer and Signalink to minimize RFI.

dancrossnyc, 2023-06-13:
Sure, closing here. However, I'm not sure about the RFI explanation; this was working fine for a few years and stopped pretty suddenly, with no hardware changes (or moves of things).

closed 2023-06-13 by dancrossnyc