wb2osz/direwolf #476: Separate transmit and receive speeds

cminyard, 2023-06-16:
I'm working on a satellite application that has different speeds on the uplink and downlink, 1200 on the uplink and 9600 on the downlink.  Is it possible to do this?  How hard would it be to add this capability?

Thanks,

-corey

wb2osz, 2023-06-16:
That is possible now.  direwolf supports multiple audio devices and multiple radio channels.  One channel can have a 1200 bps modem and another channel can have 9600.  The KISS and AGW protocols contain a field, for each packet, identifying the channel (sometimes called port) number.  The application would listen on one channel and transmit on the other.

dranch, 2023-06-16:
John, can you give some examples of what KISS or AGW-enabled applications support split channel RX/TX usage like this?

cminyard, 2023-06-16:
On Fri, Jun 16, 2023 at 06:24:13AM -0700, wb2osz wrote:
> That is possible now.  direwolf supports multiple audio devices and multiple radio channels.  One channel can have a 1200 bps modem and another channel can have 9600.  The KISS and AGW protocols [...]

Can you do it with one audio device and one radio?  Some radios can
transmit on one frequency and receive on another; it would eliminate a
lot of hardware to be able to do it that way.

Plus, many applications cannot support this.  Since we are writing the
ground station hardware we can do what you suggest, but it's a lot more
convenient to just have one connection.

Plus if we use AGW (which I would like to do), it's one AX.25
connection, transmitting on 1200 and receiving on 9600.  And it's full
duplex, you have to be able to receive while you are transmitting.  Can
direwolf do that?

Thanks for responding.  I like direwolf, I'd like to be able to use it.
If you can point me the right direction, I can possibly work on this.

-corey