wb2osz/direwolf #477: Direwolf does not respect user's terminal background colour

wyan, 2023-06-22:
The universal behaviour in any non-ncurses command line application is to only colour the text, not the background.

Direwolf sets a very bright white background on an otherwise dark background terminal, without an option to disable the background colour, other than to disable colouring text completely.

It would be greatly appreciated to have the option to have direwolf respect the background colour of the user's terminal, by not setting the background color at all.

oe3gwu, 2023-07-05:
True. I hate that white colour. Using linux, it should respect the Terminal Settings.

Gizmokid2005, 2023-08-04:
#441 addresses this, but has been untouched since created by the dev team.

LongnoseRob, 2023-08-05:
@Gizmokid2005 if you have tested #441, why not review the changes and support the dev-team in that way?

Gizmokid2005, 2023-08-05:
> @Gizmokid2005 if you have tested #441, why not review the changes and support the dev-team in that way?

Or maybe you could stop assuming and butting in with unrelated commentary on an issue you had no interaction in? I never said I tested the changes, I didn't write the changes, I don't use the software. I had an unrelated friend asking the same question, about a piece of software that had to specifically go out of its way to cause this issue in the first place, and found a pending fix as well as a related issue, and shared it to be a good steward of the tech community. You may want to look into it.

wyan, 2023-08-05:
Thank you for the tip about #441, I will try to test it and report :)

wb2osz, 2025-11-30:
This is now in the "dev" development branch and will later appear in the 1.9 release.

73,
John WB2OSZ

closed 2025-11-30 by wb2osz