wb2osz/direwolf #479: AFSK 2400 Baud

oe3gwu, 2023-07-07:
Hello!

You support QPSK2k4. That is great. But it would be nice to have AFSK2k4. Out Winlink Digi used to run UZ7HO Sound Modem using AFSK 2k4, but we migrated to Linux and use Direwolf now. We downgraded to AFSK1k2 now. Would be nice to have the 2k4 option again.

73
Rainer (OE3GWU)

dranch, 2023-07-07:
If you read https://github.com/wb2osz/direwolf/blob/master/doc/2400-4800-PSK-for-APRS-Packet-Radio.pdf , it talks about the various modes that were officially supported by TAPR modems but there is no mention of 2400bps AFSK.  I searched around and I cannot find any mention of it anywhere and everything found (legacy AEA, TNC2, etc) talks about 2400bps support was *PSK* only.

It's also worth mentioning that per http://uz7.ho.ua/modem_beta/CHANGELOG.txt :
--
SoundModem v1.05 changes:
- Added "Pi/4 QPSK V26 2400bd" AEA DSP2232, MFJ2400, DireWolf V26B compatible mode.
--

Soundmodem v1.05 was released back in December 2020 and the current version is v1.14 that was released in Oct 2022 (9 months ago).  Is there any specific reason why you cannot switch to the standard 2400bps PSK mode?

wb2osz, 2023-07-07:
(1)  Do you know what tones that UZ7HO soundmodem uses for 2400 baud AFSK?   I could not find it in the documentation or anywhere on the screen when running the application and selecting that speed.

(2) Would you consider the 2400 bps PSK mode used by some models from MFJ, AEA, and Kantronics?  I tested it with various combinations of pre-emphasis/none and de-emphasis/none and it worked great.

73,
John WB2OSZ

oe3gwu, 2023-07-08:
Maybe different in us then eu. Tnc3, scs tracker and others supported 2400 and 4800 afsk. Nobody used qpsk here.

wb2osz, 2023-07-09:
(1) Do you know what tones that UZ7HO soundmodem uses for 2400 baud AFSK? I could not find it in the documentation or anywhere on the screen when running the application and selecting that speed.

4800 bps AFSK, through voice radios does not seem possible with the limited bandwidth.

oe3gwu, 2023-09-23:
Sorry for the delay, I had some health issues the last week.

I dont know but  I try to ask him. Maybe he shares the info. Otherwise we could simply try, but I dont have a Windows Computer near me right now.

oe3gwu, 2023-09-24:
Here are some infos. I hope the help.

xtal=6.5536
lo=1774
hi=3251
center=2512
shift=1477

xtal=8.0
lo=2165
hi=3970
center=3067
shift=1805