wb2osz/direwolf #480: Use command to activate PTT

wayneworkman, 2023-07-10:
Hello, I'm using the AML-S905X-CC Libre Computer Board to run Direwolf.

I'm trying to use GPIO pin 8 for DTS and pin 10 for RTS with an EASY DIGI isolation board.

I can activate PTT on my radio by setting pin 8 high using the gpiod tools like this:
```
lgpio set 8=1
```

When trying to configure DIrewolf to use GPIO 8 for PTT, I get this error:
```
Error writing "8" to /sys/class/gpio/export, errno=22
Invalid argument
```
When I try to set GPIO 8 high manually with sysfs like such, I get this.
```
echo 8 > /sys/class/gpio/export
-bash: echo: write error: Invalid argument
```
Since the sysfs method is not working, how can I set Direwolf to issue a command for controlling PTT / RTS / DTS ? Or, is there a better way altogether to control hardware that has proper kernel drivers?

wb2osz, 2023-07-10:
This "issues" section is for reporting direwolf defects or make enhancement requests.
This is not the place to ask for help.
There are about 1800 members of the https://groups.io/g/direwolf forum.  One of them might have found a solution already.

I went looking for Libre computer documentation and could not find much.
https://docs.google.com/spreadsheets/d/1U3z0Gb8HUEfCIMkvqzmhMpJfzRqjPXq7mFLC-hvbKlE/edit#gid=0
lists many functions for GPIO 8 and it has an "*" footnote.  Perhaps it is not the best choice.  Did you try using other GPIO pins?

This was originally implemented with the sysfs approach before gpiod was available.  A future version might support gpiod approach as well.

Serial port RTS and DTR are also available for PTT.  Just put something like this in your configuration file.

PTT /dev/pttACM0 RTS

Let us know what you learn so it can be added to the documentation.

73,
john WB2OSZ

wb2osz, 2023-07-10:
If you are using a USB audio adapter, with a CMedia CM108/CM119, you can also use a GPIO pin on that chip.

PTT CM108

will automatically determine the proper HID device corresponding to the audio device.  This is covered in the Radio Interface Guide, and to a lesser extent, in the User Guide.

wayneworkman, 2023-07-11:
I will use the forums from now on for help.

I did get my issue solved. I believe I got off on the wrong foot from reading the Libre Computer forums which advise to not use sysfs for controlling GPIO pins. Though as you've stated, direwolf doesn't support gpiod and sysfs is the option available in direwolf. Knowing this, I researched and focused more on sysfs and figured out the issue.

I now know I wanted to use physical pin 8. Using the spreadsheet you linked (which is also available on the Libre Computer forums), it says a jumper needs moved to use GPIO. I moved the jumper on the board (indicated on the drawing in the spreadsheet) to the outer edge to activate the GPIO pins correctly.

Then, using the spreadsheet, you can see that physical pin 8 maps to sysfs 492.

This must first be "exported" to sysfs. The way to do that is like such:
```
echo 492 > /sys/class/gpio/export
```
Then, in direwolf.conf the PTT is defined as such:
```
PTT GPIO 492
```

Seems to be working. I'll see about contributing a more complete tutorial on the direwolf forums so the next person doesn't have so much difficulty.

closed 2023-07-11 by wayneworkman