wb2osz/direwolf #486: Version 1.7G: MEMORY LEAK, rrbb_new, new_count= Message

ngc6589, 2023-08-14:
When running direwolf 1.7G, the following message started to appear continuously. It worked for about 7-10 days. This message was displayed on the terminal running direwolf.

```
JA3VQW-10 audio level = 50(+34/-30)   [NONE]   ||||_____
[0.1] JA3VQW-10>APU25N,WIDE:}JP3YIP-I>APIRP2,TCPIP,JA3VQW-10*:!3500.36ND13546.00E&D-STAR -> APRS<0x0d>
Position, I=Igte IGate R=RX T=1hopTX 2=2hopTX w/overlay D
N 35 00.3600, E 135 46.0000
D-STAR -> APRS
Rx IGate: Unwrap third party message.
Source SSID must be digits. "JP3YIP-I" has letters in SSID.
Failed to create packet from text.  Bad source address

JA3VQW-10 audio level = 54(+43/-30)   [NONE]   ___:|:___
[0.4] JA3VQW-10>APU25N,WIDE:}JP3YIP-I>APIRP2,TCPIP,JA3VQW-10*:<IGATE,MSG_CNT=0,LOC_CNT=1<0x0d>
Station Capabilities
IGATE,MSG_CNT=0,LOC_CNT=1
Rx IGate: Unwrap third party message.
Source SSID must be digits. "JP3YIP-I" has letters in SSID.
Failed to create packet from text.  Bad source address
MEMORY LEAK, rrbb_new, new_count=70453630, delete_count=70453529
MEMORY LEAK, rrbb_new, new_count=70453631, delete_count=70453530
MEMORY LEAK, rrbb_new, new_count=70453632, delete_count=70453531
MEMORY LEAK, rrbb_new, new_count=70453633, delete_count=70453532
MEMORY LEAK, rrbb_new, new_count=70453634, delete_count=70453533
MEMORY LEAK, rrbb_new, new_count=70453635, delete_count=70453534
MEMORY LEAK, rrbb_new, new_count=70453636, delete_count=70453535
MEMORY LEAK, rrbb_new, new_count=70453637, delete_count=70453536
MEMORY LEAK, rrbb_new, new_count=70453638, delete_count=70453537
MEMORY LEAK, rrbb_new, new_count=70453639, delete_count=70453538
MEMORY LEAK, rrbb_new, new_count=70453640, delete_count=70453539
MEMORY LEAK, rrbb_new, new_count=70453641, delete_count=70453540
MEMORY LEAK, rrbb_new, new_count=70453642, delete_count=70453541
MEMORY LEAK, rrbb_new, new_count=70453643, delete_count=70453542
MEMORY LEAK, rrbb_new, new_count=70453644, delete_count=70453543
MEMORY LEAK, rrbb_new, new_count=70453645, delete_count=70453544
MEMORY LEAK, rrbb_new, new_count=70453646, delete_count=70453545
MEMORY LEAK, rrbb_new, new_count=70453647, delete_count=70453546
MEMORY LEAK, rrbb_new, new_count=70453648, delete_count=70453547
MEMORY LEAK, rrbb_new, new_count=70453649, delete_count=70453548
MEMORY LEAK, rrbb_new, new_count=70453650, delete_count=70453549
MEMORY LEAK, rrbb_new, new_count=70453651, delete_count=70453550
MEMORY LEAK, rrbb_new, new_count=70453652, delete_count=70453551
MEMORY LEAK, rrbb_new, new_count=70453653, delete_count=70453552
MEMORY LEAK, rrbb_new, new_count=70453654, delete_count=70453553
MEMORY LEAK, rrbb_new, new_count=70453655, delete_count=70453554
MEMORY LEAK, rrbb_new, new_count=70453656, delete_count=70453555
MEMORY LEAK, rrbb_new, new_count=70453657, delete_count=70453556
MEMORY LEAK, rrbb_new, new_count=70453658, delete_count=70453557
MEMORY LEAK, rrbb_new, new_count=70453659, delete_count=70453558
MEMORY LEAK, rrbb_new, new_count=70453660, delete_count=70453559
```

There were no memory leak messages in the log file specified in direwolf.conf.

```
0,1690910137,2023-08-01T17:15:37Z,JJ3CYE-10,JA3VQW-10,55(+36/-29),0,},JJ3CYE-10,/ ,,,,,,,,,UIview 32 bit apps,,,430.980MNz T77 Kansa
i Flower Musen Club
0,1690910138,2023-08-01T17:15:38Z,JA8PGJ-10,JA3VQW-10,53(+34/-29),0,},JA8PGJ-10,I&,42.966667,140.513167,,,,144.660,,,UIview 32 bit a
pps,,,Iwanai Town I-Gate144.660 1200bps {UIV32N}
0,1690910138,2023-08-01T17:15:38Z,JP3YIP-I,JA3VQW-10,50(+34/-30),0,},JP3YIP-I,D&,35.006000,135.766667,,,,,,,,,,D-STAR -> APRS
0,1690910138,2023-08-01T17:15:38Z,JP3YIP-I,JA3VQW-10,54(+43/-30),0,},JP3YIP-I,/ ,,,,,,,,,,,,"IGATE,MSG_CNT=0,LOC_CNT=1"
2,1690910150,2023-08-01T17:15:50Z,JR3XMG-1,JR3XMG-1,63(16/8),0,U,JR3XMG-1,/ ,,,,,,,,,,,,
0,1690910200,2023-08-01T17:16:40Z,JH1HFE,JA3VQW-10,55(+35/-31),1,},JH1HFE-8,\P,42.068833,140.773333,,,,,,,SV2AGW's AGWtracker,,,
2,1690910205,2023-08-01T17:16:45Z,JR3XMG-1,JR3XMG-1,63(16/8),0,!,JR3XMG-1,1#,34.319000,135.158833,,,,144.660,,,UIdigi,,,144.66MHz 1200bps Misaki
2,1690910235,2023-08-01T17:17:15Z,7J3ACR-10,7J3ACR-10,51(11/7),0,=,7J3ACR-10,I&,34.685002,135.176669,,,,,,,UIview 32 bit apps,,,HarborLand Kobe Japan {UIV32N}
0,1690910259,2023-08-01T17:17:39Z,JJ1LFO-10,JA3VQW-10,56(+36/-30),0,},JJ1LFO-10,/ ,,,,,,,,,KA2DDO Yet another APRS system,,,PHG20202/
2,1690910263,2023-08-01T17:17:43Z,7J3ACR-10,7J3ACR-10,60(10/7),0,>,7J3ACR-10,/ ,,,,,,,,,UIview 32 bit apps,,,UI-View32 V2.03
2,1690910273,2023-08-01T17:17:53Z,7J3ACR-10,7J3ACR-10,61(12/7),0,<,7J3ACR-10,/ ,,,,,,,,,UIview 32 bit apps,,,"IGATE,MSG_CNT=0,LOC_CNT=0"
2,1690910303,2023-08-01T17:18:23Z,7J3ACR-10,7J3ACR-10,59(10/7),0,<,7J3ACR-10,/ ,,,,,,,,,UIview 32 bit apps,,,"IGATE,CLOSED"
0,1690910322,2023-08-01T17:18:42Z,JG6YCL-13,JA3VQW-10,54(+38/-30),1,},JG6YCL-13,/_,33.803000,130.565833,,,,,,,"KetaiTracker by  JF6LZE, Takeki (msg capable)",,,
0,1690910326,2023-08-01T17:18:46Z,JH5HOA-10,JA3VQW-10,56(+36/-30),0,},JH5HOA-10,/ ,,,,,,,,,UIview 32 bit apps,,,"IGATE,MSG_CNT=0,LOC_CNT=3"
0,1690910326,2023-08-01T17:18:46Z,JG5MMU-10,JA3VQW-10,52(+33/-31),1,},JG5MMU-10,I&,34.093498,132.975497,,,,,,,UIview 32 bit apps,,,Imabari-city Ehime Japan {UIV32N}
0,1690910381,2023-08-01T17:19:41Z,JP4YDD-I,JA3VQW-10,55(+35/-30),0,},JP4YDD-I,D&,35.238000,134.085167,,,,,,,,,,D-STAR -> APRS
2,1690910423,2023-08-01T17:20:23Z,7J3ACR-10,7J3ACR-10,56(11/7),0,<,7J3ACR-10,/ ,,,,,,,,,UIview 32 bit apps,,,"IGATE,MSG_CNT=0,LOC_CNT=0"
0,1690910441,2023-08-01T17:20:41Z,JH5EWP-10,JA3VQW-10,54(+37/-30),1,},JH5EWP-10,/ ,,,,,,,,,UIview 32 bit apps,,,"IGATE,MSG_CNT=0,LOC:
```
The execution environment is Raspberry Pi 4, 64-bit environment.
direwolf uses git cloned develop branch. and attach a .conf file.
```
$ uname -a
Linux pi4-aprs 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
```

```
LOGDIR /home/pi/d1296log
ADEVICE  plughw:1,0
ARATE 48000
ADEVICE1 plughw:0,0
ARATE 48000
CHANNEL 0
MYCALL JP3SRS-10
MODEM 9600
PTT /dev/ttyUSB0 RTS
TXDELAY 40
CHANNEL 2
MODEM 1200
PTT /dev/ttyUSB1 RTS

AGWPORT 8000
KISSPORT 8001
KISSPORT 0

DIGIPEAT 0 0 ^WIDE[3-7]-[1-7]$|^TEST$ ^WIDE[12]-[12]$ TRACE
DIGIPEAT 2 2 ^WIDE[3-7]-[1-7]$|^TEST$ ^WIDE[12]-[12]$ TRACE

IGSERVER japan.aprs2.net
IGLOGIN JP3SRS-10 123456789

PBEACON SENDTO=0 DELAY=1 EVERY=15:00 OBJNAME=JP3SRS-10 SYMBOL=\& OVERLAY=I LAT=34.656720 LONG=135.646533 COMMENT="Dire Wolf version 1.7 G for Raspberry Pi."
PBEACON SENDTO=2 DELAY=2 EVERY=15:00 OBJNAME=JP3SRS-10 SYMBOL=\& OVERLAY=I LAT=34.656720 LONG=135.646533 COMMENT="Dire Wolf version 1.7 G for Raspberry Pi"

IGTXVIA 0 WIDE1-1
IGTXVIA 2 WIDE1-1

APRStt GATEWAY remains a sample
```
`direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii`

I don't know the conditions to reproduce, and I'm building without the Debug option, so the only thing I can report as a problem is that the message occurred.

73 Masahiro Kusunoki JP3SRS

wb2osz, 2023-08-16:
Thank you for reporting this.
I have not been able to reproduce this issue but I think commit  7a8e432 will solve the problem.
Please try the most recent, in the dev branch, and let me know how it goes.

73, John WB2OSZ

ngc6589, 2023-08-16:
WB2OSZ

Hello.
Thanks for the fix. I built commit number 7a8e432 today and restarted direwolf.
Let it run for 2 weeks and see if the Memory Leak message occurs.

73 JP3SRS

F4FXL, 2023-08-22:
Hi,

I had similar issue on 3 of my setups.
- One system is running Armbian Bullsseye 32bits
- Two systems running raspbian 64bits

Now testing with the fix.

F4FXL, 2023-08-23:
Only one channel.

Direwolf crashed and restarted after approx 10 to 15 days. Yesterday I noticed that one of my digipeaters stopped sending its beacons to RF but was sending them to IS. I think it had just reached some erratic state before crashing. I checked the other digipeaters and noticed they had also restarted.

I am running Direwold as a [systemd ](https://www.f4fxl.org/start-direwolf-at-boot-the-systemd-way/) service through tmux so the terminal output is lost when the service gets restarted upon crash.

ngc6589, 2023-08-24:
I was doing a direwolf running test. It ended in a week due to a power outage. Lightning is the cause.

Here is the output of the ps aux -L command after a week.
```
USER       PID   LWP %CPU NLWP %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
pi       21526 21526  0.0   13  0.1 119288  5460 pts/1    Sl+  Aug17   1:26 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi       21526 21528  0.0   13  0.1 119288  5460 pts/1    Sl+  Aug17   0:09 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi       21526 21529  0.0   13  0.1 119288  5460 pts/1    Sl+  Aug17   0:26 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi       21526 21530  0.0   13  0.1 119288  5460 pts/1    Sl+  Aug17   0:05 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi       21526 21531  0.0   13  0.1 119288  5460 pts/1    Sl+  Aug17   0:25 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi       21526 21532  0.0   13  0.1 119288  5460 pts/1    Sl+  Aug17   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi       21526 21533  0.0   13  0.1 119288  5460 pts/1    Sl+  Aug17   0:21 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi       21526 21534  0.0   13  0.1 119288  5460 pts/1    Sl+  Aug17   0:23 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi       21526 21535  0.0   13  0.1 119288  5460 pts/1    Sl+  Aug17   0:22 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi       21526 21536  0.0   13  0.1 119288  5460 pts/1    Sl+  Aug17   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi       21526 21537  0.0   13  0.1 119288  5460 pts/1    Sl+  Aug17   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi       21526 21538 27.3   13  0.1 119288  5460 pts/1    Rl+  Aug17 2748:13 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi       21526 21539 15.2   13  0.1 119288  5460 pts/1    Sl+  Aug17 1536:52 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
```
Usage of aprs Raspberry Pi collected by cacti was stable.(A graph for one week is output)
[image]

I couldn't check for 2 weeks due to the power outage.
Since direwolf has restarted, I would like to see the progress again.
I think that the current situation is stable, so I will leave the Close judgment to you.
(Regardless of open/close, the result will be written here again)

ngc6589, 2023-08-31:
I checked the VSS RSS in the ps command after starting direwolf. There was no change in VSS RSS in one week. The system is running stable.
The cacti graph is also stable with no change in CPU usage.

I think the volatile setting of the counter variable you described worked. Personally, I think it is fine to close it.

73 JP3SRS
'''

2023/08/24 restart direwolf (Power outage due to thunderStorm)
USER       PID   LWP %CPU NLWP %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
pi         969   969  0.0   13  0.1 116216  4904 pts/1    Sl+  14:34   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   971  0.0   13  0.1 116216  4904 pts/1    Sl+  14:34   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   972  0.0   13  0.1 116216  4904 pts/1    Sl+  14:34   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   973  0.0   13  0.1 116216  4904 pts/1    Sl+  14:34   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   974  0.0   13  0.1 116216  4904 pts/1    Sl+  14:34   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   975  0.0   13  0.1 116216  4904 pts/1    Sl+  14:34   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   976  0.0   13  0.1 116216  4904 pts/1    Sl+  14:34   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   977  0.0   13  0.1 116216  4904 pts/1    Sl+  14:34   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   978  0.0   13  0.1 116216  4904 pts/1    Sl+  14:34   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   979  0.0   13  0.1 116216  4904 pts/1    Sl+  14:34   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   980  0.0   13  0.1 116216  4904 pts/1    Sl+  14:34   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   981 25.5   13  0.1 116216  4904 pts/1    Rl+  14:34   0:10 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   982 14.1   13  0.1 116216  4904 pts/1    Rl+  14:34   0:06 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii

2023/08/24 15:00
pi@pi4-aprs:~ $ ps aux -L | grep dire
USER       PID   LWP %CPU NLWP %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
pi         969   969  0.0   13  0.1 119288  5684 pts/1    Sl+  14:34   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   971  0.0   13  0.1 119288  5684 pts/1    Sl+  14:34   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   972  0.0   13  0.1 119288  5684 pts/1    Sl+  14:34   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   973  0.0   13  0.1 119288  5684 pts/1    Sl+  14:34   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   974  0.0   13  0.1 119288  5684 pts/1    Sl+  14:34   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   975  0.0   13  0.1 119288  5684 pts/1    Sl+  14:34   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   976  0.0   13  0.1 119288  5684 pts/1    Sl+  14:34   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   977  0.0   13  0.1 119288  5684 pts/1    Sl+  14:34   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   978  0.0   13  0.1 119288  5684 pts/1    Sl+  14:34   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   979  0.0   13  0.1 119288  5684 pts/1    Sl+  14:34   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   980  0.0   13  0.1 119288  5684 pts/1    Sl+  14:34   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   981 27.2   13  0.1 119288  5684 pts/1    Sl+  14:34   9:01 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   982 15.2   13  0.1 119288  5684 pts/1    Rl+  14:34   5:04 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii

2023/08/25 08:10
pi@pi4-aprs:~ $ ps aux -L | grep dire
USER       PID   LWP %CPU NLWP %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
pi         969   969  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:05 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   971  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   972  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:01 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   973  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   974  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:02 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   975  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   976  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:02 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   977  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:02 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   978  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:02 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   979  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   980  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
[... cut ...]
pi         969   979  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   980  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   981 27.2   13  0.1 119288  5684 pts/1    Sl+  Aug24 2702:48 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   982 15.3   13  0.1 119288  5684 pts/1    Sl+  Aug24 1521:29 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii

'''
[image]

ngc6589, 2023-09-08:
To follow up on one previous message, it has been 2 weeks since I started direwolf.
It is running without any problems.

```
2023/09/02 07:56
pi@pi4-aprs:~ $ ps aux -L | grep dire
pi         969   969  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   1:14 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   971  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:10 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   972  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:14 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   973  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:06 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   974  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:25 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   975  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   976  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:26 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   977  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:28 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   978  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:27 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   979  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   980  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   981 27.2   13  0.1 119288  5684 pts/1    Rl+  Aug24 3418:56 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   982 15.3   13  0.1 119288  5684 pts/1    Sl+  Aug24 1924:30 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii

2023/09/03 07:55
pi@pi4-aprs:~ $ ps aux -L | grep dire
pi         969   969  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   1:24 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   971  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:12 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   972  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:16 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   973  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:07 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   974  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:28 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   975  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   976  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:29 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   977  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:31 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   978  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:30 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   979  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   980  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   981 27.2   13  0.1 119288  5684 pts/1    Sl+  Aug24 3810:18 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   982 15.3   13  0.1 119288  5684 pts/1    Sl+  Aug24 2144:57 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii

2023/09/04 08:19
pi@pi4-aprs:~ $ ps aux -L | grep  dire
pi         969   969  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   1:32 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   971  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:13 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   972  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:18 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   973  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:08 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   974  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:31 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   975  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   976  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:33 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   977  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:34 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   978  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:33 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   979  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   980  0.0   13  0.1 119288  5684 pts/1    Sl+  Aug24   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   981 27.2   13  0.1 119288  5684 pts/1    Sl+  Aug24 4209:49 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   982 15.3   13  0.1 119288  5684 pts/1    Sl+  Aug24 2369:55 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii

2023/09/05 09:03
pi@pi4-aprs:~ $ ps aux -L | grep dire
pi         969   969  0.0   13  0.1 119288  5944 pts/1    Sl+  Aug24   1:40 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
[... cut ...]
pi         969   980  0.0   13  0.1 119288  5944 pts/1    Sl+  Aug24   0:00 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   981 27.2   13  0.1 119288  5944 pts/1    Sl+  Aug24 5849:15 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
pi         969   982 15.3   13  0.1 119288  5944 pts/1    Sl+  Aug24 3292:35 direwolf -c d1296.conf -r 48000 -b 16 -d fff -d ii
```

The cacti graphs for the two weeks were also very stable.

[image]

No more memory leak messages in my environment. Thank you very much.

DE JP3SRS Masahiro Kusunoki

wb2osz, 2023-09-08:
Thank you for all of your testing and analysis.
I don't think there was actually a memory leak; the counters were not right due to multiple threads updating them.
I think we can close this issue now.

closed 2023-09-08 by wb2osz