wb2osz/direwolf #488: digipath: WIDE1-1,WIDE2-1 issue

hwsu315, 2023-09-11:
Dear OM,

I found when using direwolf as the wide range digipeater which only repeat WIDE2 packets, the direwolf doesn't repeat WIDE1-1,WIDE2-1 packets.

In other words, for trackers with WIDE1-1,WIDE2-1 beacons and there are no fill-in digis surrounding by, the wide range direwolf digi hears it directly but does not repeat.

direwolf digipeater settings:
DIGIPEAT 0 0 ^WIDE[3-7]-[1-7]$ ^WIDE2-[12]$ TRACE

As far as I understand, wide range digipeaters should repeat WIDE1-1,WIDE2-1 ; WIDE2-N ; but not WIDE1-1.

Thanks for investigating the issue.
Regrards,
Huawei Su 73 de BX2AI

hwsu315, 2023-09-20:
Any updates about this issue?

dranch, 2023-09-20:
Can you tell us about your setup?  What version of Direwolf are you running?  Can you show us some examples say give us a URL from aprs.fi RAW view so we can see some of the packets?

wb2osz, 2023-09-20:
It is working properly. 
A digipeater only examines the **first unused** digipeater address in the via path.
In this case, the digipeater only looks at WIDE1-1 and the configuration does not include anything to process WIDE1-n.
You might find this useful:  https://github.com/wb2osz/direwolf-doc/blob/main/APRS-Digipeaters.pdf 
There is an obscure digipeater option called **preemptive** digipeating which would probably help in this situation.   I will make an example and share with you.

hwsu315, 2023-09-21:
Thanks a lot for the info, I just start to study the "preemptive digipeating" function. I will go to read user-guide and try it. Regards, Huawei.
73 de BX2AI

na7q, 2023-10-27:
You have a typo in your settings.
Fix it with this at the tail end where you are missing the 1

 ^WIDE[12]-[12]