wb2osz/direwolf #489: Linux glibc 2.38 implements strlcat/strlcpy

adrians, 2023-09-20:
As per the release-announcement for glibc 2.38 (<https://lists.gnu.org/archive/html/info-gnu/2023-07/msg00010.html>) :
```
* The strlcat and strlcpy functions have been added.  They are derived
  from OpenBSD, and are expected to be added to a future POSIX version.
```

When doing Linux builds using the updated system library, the `strlcat`/`strlcpy` implementations from glibc clash with the implementations from this project (referenced in `src/direwolf.h`, `external/misc/strlcat.c` and `external/misc/strlcpy.c`).

More precisely, when compiling `strlcat.c`, the compiler first saw the preprocessor-define from `src/direwolf.h:301`, tried redefining all instances of `strlcpy` to `strlcpy_debug`, and then while resolving other include-directives started redefining the `strlcpy` references from glibc to point to `strlcpy_debug`.

The error message reported by gcc was about `src/direwolf.h:301,302`, not explaining too much about the root cause:
```
expected declaration specifiers or '...' before string constant
```

While for my purposes I've fixed this bug (the project now compiles under Linux, patch [https://github.com/adrians/meta-amateurradio/blob/master/recipes-ham/direwolf/files/0001-disable-strlcat-strlcpy-implementations.patch](here)), I'm also reporting it upstream to raise awareness about this issue (hopefully no-one else wastes time trying to understand the error message in this scenario).

sarcasticadmin, 2023-10-29:
nixpkgs build of `direwolf` ran into this as well and fixed it with a patch: https://github.com/NixOS/nixpkgs/pull/247401/commits/d00a220853e808e2fcc9b951628645e779d56db6

Adding it hear so theres more visibility of this issue across the community

wb2osz, 2023-10-29:
Excerpt from CMakeLists.txt:

# Some platforms provide their own strlcpy & strlcat. (BSD, MacOSX)
# Others don't so we provide our own. (Windows, most, but not all Linux)
# Here we detect whether these are provided by the OS and set a symbol
# so that:
#  (1) libgps does not supply its own version.
#  (2) we know whether we need to supply our own copy.
#
# This was all working fine until these were added to the gnu c library 2.38.
# References:
#  - https://www.gnu.org/software/libc/sources.html
#  - https://sourceware.org/git/?p=glibc.git;a=blob_plain;f=NEWS;hb=HEAD
#
# This test is not detecting them for glibc 2.38 resulting in a conflict.
# Why?  Are they declared in a different file or in some strange way?
#
# This is how they are declared in include/string.h:
#
#       extern __typeof (strlcpy) __strlcpy;
#       libc_hidden_proto (__strlcpy)
#       extern __typeof (strlcat) __strlcat;
#       libc_hidden_proto (__strlcat)
#
# Apparently cmake does not recognize this style.
# Keep this here for BSD type systems where it behaves as expected.
# We will need to add a hack in direwolf.h to define these if glibc version >= 2.38.

wb2osz, 2023-10-29:
I have not been able to test this yet when starting with the most recent Ubuntu.
I couldn't find a package to install glibc 2.38 and when I tried to install from source, the system got trashed and became unusable.

NixOS makes no sense to me.  I figured out how to effectively install applications (e.g. gcc) with "nix-shell -p ..." but could not figure how to get the required libraries.

Does anyone have some suggestions on how to set up a test OS with this new library version?

sarcasticadmin, 2023-10-29:
> Does anyone have some suggestions on how to set up a test OS with this new library version?

@wb2osz I would do the following to get a build environment for direwolf with nix:

```
$ nix develop nixpkgs#direwolf # To get a new shell with the build environment for direwolf
$ ldd --version # To confirm youve got the right glibc
```

Then you can go about building direwolf per the normal direwolf build instructions in this repo.

If you'd like to know more about `nix develop` checkout the manual for it here: https://nixos.org/manual/nix/stable/command-ref/new-cli/nix3-develop

One thing I did notice was that while this was broken on `1.6` and required the patch I mentioned https://github.com/NixOS/nixpkgs/commit/d00a220853e808e2fcc9b951628645e779d56db6

This breakage seems to have been fixed in the most recent version `1.7`:

```
driver direwolf $ mkdir build
driver direwolf $ cd build/
driver build $ cmake ..
-- The C compiler identification is GNU 12.3.0
-- The CXX compiler identification is GNU 12.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /nix/store/zlzz2z48s7ry0hkl55xiqp5a73b4mzrg-gcc-wrapper-12.3.0/bin/gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /nix/store/zlzz2z48s7ry0hkl55xiqp5a73b4mzrg-gcc-wrapper-12.3.0/bin/g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /run/current-system/sw/bin/git (found version "2.40.1")
-- Dire Wolf Version: 1.7.0-de293a1
-- Build type set to: Release
CMake system: Linux
-- Target architecture: x86_64
-- Use SSE SIMD instructions
-- Looking for strlcpy
-- Looking for strlcpy - found
-- Looking for strlcat
-- Looking for strlcat - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Found hamlib: /nix/store/dpj7dnbkqljwg3gmcqzwrv1x966bkqb1-hamlib-4.5.5/lib/libhamlib.so
-- Found ALSA: /nix/store/4vpj5g0cprbybn6k29rzf1108az9a1ii-alsa-lib-1.2.9/lib/libasound.so (found version "1.2.9")
-- Found udev: /nix/store/vapnrxrw3b21c7ji61bmnbzl2cj1vl96-systemd-254.3/lib/libudev.so
-- Could NOT find Avahi (missing: AVAHI_COMMON_FOUND AVAHI_CLIENT_FOUND)
-- Configuring done (0.7s)
-- Generating done (0.0s)
-- Build files have been written to: /home/rherna/workspace/direwolf/build
driver build $ make
[  0%] Building C object external/geotranz/CMakeFiles/geotranz.dir/error_string.c.o
[  1%] Building C object external/geotranz/CMakeFiles/geotranz.dir/mgrs.c.o
[  1%] Building C object external/geotranz/CMakeFiles/geotranz.dir/polarst.c.o
[  2%] Building C object external/geotranz/CMakeFiles/geotranz.dir/tranmerc.c.o
[  2%] Building C object external/geotranz/CMakeFiles/geotranz.dir/ups.c.o
[  3%] Building C object external/geotranz/CMakeFiles/geotranz.dir/usng.c.o
[  3%] Building C object external/geotranz/CMakeFiles/geotranz.dir/utm.c.o
[  4%] Linking C static library libgeotranz.a
[  4%] Built target geotranz
[  5%] Building C object src/CMakeFiles/direwolf.dir/direwolf.c.o
[  5%] Building C object src/CMakeFiles/direwolf.dir/ais.c.o
[  6%] Building C object src/CMakeFiles/direwolf.dir/aprs_tt.c.o
[  6%] Building C object src/CMakeFiles/direwolf.dir/audio_stats.c.o
[  7%] Building C object src/CMakeFiles/direwolf.dir/ax25_link.c.o
/home/rherna/workspace/direwolf/src/ax25_link.c: In function ‘dl_connect_request’:
/home/rherna/workspace/direwolf/src/ax25_link.c:975:13: warning: ‘get_link_handle’ accessing 120 bytes in a region of size 12 [-Wstringop-overflow=]
[... 257 lines cut ...]
[ 91%] Building C object src/CMakeFiles/tnctest.dir/textcolor.c.o
[ 92%] Building C object src/CMakeFiles/tnctest.dir/dtime_now.c.o
[ 92%] Building C object src/CMakeFiles/tnctest.dir/serial_port.c.o
[ 93%] Linking C executable tnctest
[ 93%] Built target tnctest
[ 93%] Building C object src/CMakeFiles/cm108.dir/cm108.c.o
[ 94%] Building C object src/CMakeFiles/cm108.dir/textcolor.c.o
[ 94%] Linking C executable cm108
[ 94%] Built target cm108
[ 94%] Building C object src/CMakeFiles/ttcalc.dir/ttcalc.c.o
[ 95%] Building C object src/CMakeFiles/ttcalc.dir/ax25_pad.c.o
[ 95%] Building C object src/CMakeFiles/ttcalc.dir/fcs_calc.c.o
[ 96%] Building C object src/CMakeFiles/ttcalc.dir/textcolor.c.o
[ 96%] Linking C executable ttcalc
[ 96%] Built target ttcalc
[ 96%] Building C object src/CMakeFiles/appserver.dir/appserver.c.o
[ 97%] Building C object src/CMakeFiles/appserver.dir/agwlib.c.o
[ 97%] Building C object src/CMakeFiles/appserver.dir/dwsock.c.o
[ 98%] Building C object src/CMakeFiles/appserver.dir/dtime_now.c.o
[ 98%] Building C object src/CMakeFiles/appserver.dir/ax25_pad.c.o
[ 99%] Building C object src/CMakeFiles/appserver.dir/fcs_calc.c.o
[ 99%] Building C object src/CMakeFiles/appserver.dir/textcolor.c.o
[100%] Linking C executable appserver
[100%] Built target appserver
```

sarcasticadmin, 2023-10-31:
@wb2osz This longer seems like an issue for me

wb2osz, 2023-11-06:
Thank you for confirming that direwolf release 1.7 seems to be OK with glibc 2.38 on  NixOS.
I'm leaving this open until I can do some of my own testing.

adrians, 2023-11-07:
I'm also confirming that release 1.7 works with glibc 2.38 on poky (yocto build system).

wb2osz, 2024-08-02:
This was fixed in the direwolf "dev" branch which will become release 1.8 at some point.