wb2osz/direwolf #490: dev branch does not want to work without config file

LongnoseRob, 2023-10-06:
I build DW on alpine yesterday from the dev-branch, while i had no issues with building it on arm(32bit), there seems to be an issue with the config:
 I just wanted to quickly test DW from the shell without config file, I get the following messge:
```
$ direwolf -n 1 -r 48000 -b 16 udp:7355 
Dire Wolf version 1.7 (Oct  5 2023) BETA TEST 7
Includes optional support for:  dns-sd
ERROR - Could not open config file /home/xyz/direwolf.conf
Try using -c command line option for alternate location.
See online documentation:
    stable release:    https://github.com/wb2osz/direwolf/tree/master/doc
    development version:    https://github.com/wb2osz/direwolf/tree/dev/doc
    additional topics:    https://github.com/wb2osz/direwolf-doc

```
Adding a  `-B 1200` does not change anything.

Is this behaviour intended?

Same probnlem also on a slighrtly older build on my desktop:
```
$ direwolf -n 1 -r 48000 -b 16 udp:7355 
Dire Wolf DEVELOPMENT version 1.7 G (Jun 25 2023)
Includes optional support for:  hamlib cm108-ptt dns-sd
ERROR - Could not open config file /home/xyz/direwolf.conf
Try using -c command line option for alternate location.
See online documentation:
    stable release:    https://github.com/wb2osz/direwolf/tree/master/doc
    development version:    https://github.com/wb2osz/direwolf/tree/dev/doc
    additional topics:    https://github.com/wb2osz/direwolf-doc
```

wb2osz, 2023-10-06:
It really wants a configuration file.  This is intentional.
Two workarounds would be:
(1) Create an empty file.   "touch direwolf.conf"  or
(2) Specify the null device.  "direwolf -c /dev/null"

73,
John WB2OSZ

LongnoseRob, 2023-10-06:
Understood, 
Thank you, 73

closed 2023-10-06 by LongnoseRob