wb2osz/direwolf #497: Unable to establish connections use AGW

larsks, 2023-11-01:
I'm seeing some odd behavior trying to connect to remote systems via Direwolf's AGW support.

Originally I was writing my own code using the [`wl2k-go` package](https://github.com/la5nta/wl2k-go), and I thought maybe the problem was either with that package or with my code...but I'm able to reproduce the same behavior using [`agwterm`](https://www.soundcardpacket.org/6AGWPrograms.aspx).

The issue is that while connections are established successfully, Direwolf shuts down the connection as soon as the remote host sends an RR frame. That is:

1. Local system sends SABME/SABM frame
1. Remote system sends UA frame
1. Remote system sends several I frames with content (these are received correctly)
1. Remote system sends RR frame
1. Local systems replies with XID REJECT frame
1. Local system replies tall all subsequent frames with DM-

I've been testing things against a local packet BBS and a local digipeater. For example, here's the log from attempting to contact a local packet BBS using `agwterm`:

```
Attempting connect to KZ2X-1 ...
[0L] N1LKS>KZ2X-1:(SABME cmd, p=1)
N1LKS audio level = 35(21/19)    334______
[0.1] N1LKS>KZ2X-1:(SABME cmd, p=1)
KZ2X-1 audio level = 103(45/41)    0000001__
[0.3] KZ2X-1>N1LKS:(UA res, f=1)
Stream 14: Connected to KZ2X-1.  (v2.2)
KZ2X-1 audio level = 104(43/41)    0000000__
[0.3] KZ2X-1>N1LKS:(I cmd, n(s)=0, n(r)=0, p=0, pid=0xf0)CONNECTED TO KZ2X TIP #1 ON AX.25 FROM N1LKS<0x0d><0x0d>TYPE 'HOSTS' FOR HOSTS, 'EXIT' TO LOGOUT, 'HELP' FOR HELP<0x0d>TYPE 'CONNECT HOSTNAME' TO CONNECT TO A HOST<0x0d><0x0d>Pad><0x20>
[0L] N1LKS>KZ2X-1:(XID cmd, p=1) Half-Duplex REJ SREJ Multi-SREJ modulo-128 I-Field-Length-Rx=256 Window-Size-Rx=32 Ack-Timer=3000 Retries=10
[0L] N1LKS>KZ2X-1:(DM res, f=1)
[0L] N1LKS>KZ2X-1:(DM res, f=1)
```

A connection using the kernel AX.25 support (`axcall radio KZ2X-1`) works just fine:

```
[0L] N1LKS>KZ2X-1:(SABM cmd, p=1)
[0L] N1LKS>KZ2X-1:(SABM cmd, p=1)
N1LKS audio level = 34(29/20)    0246_____
[0.0] N1LKS>KZ2X-1:(SABM cmd, p=1)
[0L] N1LKS>KZ2X-1:(SABM cmd, p=1)
N1LKS audio level = 35(19/18)    011______
[0.0] N1LKS>KZ2X-1:(SABM cmd, p=1)
KZ2X-1 audio level = 103(44/40)    0000000__
[0.3] KZ2X-1>N1LKS:(UA res, f=1)
KZ2X-1 audio level = 102(43/41)    0000002__
[0.3] KZ2X-1>N1LKS:(I cmd, n(s)=0, n(r)=0, p=0, pid=0xf0)CONNECTED TO KZ2X TIP #1 ON AX.25 FROM N1LKS<0x0d><0x0d>TYPE 'HOSTS' FOR HOSTS, 'EXIT' TO LOGOUT, 'HELP' FOR HELP<0x0d>TYPE 'CONNECT HOSTNAME' TO CONNECT TO A HOST<0x0d><0x0d>Pad><0x20>
[0L] N1LKS>KZ2X-1:(RR res, n(r)=1, f=0)
N1LKS audio level = 35(18/21)    73_______
[0.1] N1LKS>KZ2X-1:(RR res, n(r)=1, f=0)
[0L] N1LKS>KZ2X-1:(I cmd, n(s)=0, n(r)=1, p=0, pid=0xf0)help<0x0d>
N1LKS audio level = 36(17/21)    2________
[0.0] N1LKS>KZ2X-1:(I cmd, n(s)=0, n(r)=1, p=0, pid=0xf0)help<0x0d>
KZ2X-1 audio level = 103(44/40)    |000000__
[0.3] KZ2X-1>N1LKS:(I cmd, n(s)=1, n(r)=1, p=0, pid=0xf0)<0x0d>
KZ2X-1 audio level = 102(44/39)    1000000__
[0.3] KZ2X-1>N1LKS:(I cmd, n(s)=2, n(r)=1, p=1, pid=0xf0)<0x0d>TIP COMMANDS:<0x0d><0x0d>HELP    - SHOW TIP HELP<0x0d>HOSTS   - DISPLAY KNOWN HOSTS<0x0d>EXIT    - END TIP SESSION AND DISCONNECT<0x0d>CONNECT - CONNECT TO A TIP HOST<0x0d><0x0d>Pad><0x20>
.
.
.
```

If I configure the remote system as AX.25 V2.0 only by setting:

```
V20 KZ2X-1
V20 BROCK
```

Then I see a similar problem. There's no XID response, but we immediately start replying with DM frames after receiving data from the remote host:

```
Attempting connect to BROCK ...
[0L] N1LKS>BROCK:(SABM cmd, p=1)
N1LKS audio level = 35(18/17)    _2_______
[0.1] N1LKS>BROCK:(SABM cmd, p=1)
BROCK audio level = 98(44/39)   [NONE]   |||||||__
[0.3] BROCK>N1LKS:(UA res, f=1)
Stream 1: Connected to BROCK.  (v2.0)
BROCK audio level = 99(45/40)   [NONE]   _||||||__
[0.3] BROCK>N1LKS:(I cmd, n(s)=0, n(r)=0, p=0, pid=0xf0)###CONNECTED TO NODE BROCK(W1MV-7) CHANNEL A<0x0d>Welcome to BROCK (W1MV-7) in Brockton, Mass<0x0d>ENTER COMMAND: B,C,J,N, or Help ?<0x0d>
[0L] N1LKS>BROCK:(DM res, f=1)
[0L] N1LKS>BROCK:(DM res, f=1)
[0L] N1LKS>BROCK:(RR res, n(r)=1, f=0)
N1LKS audio level = 2(9/8)    8________
[0.0] N1LKS>BROCK:(DM res, f=1)
N1LKS audio level = 1(7/6)    __5______
[0.2] N1LKS>BROCK:(RR res, n(r)=1, f=0)
Warning: Client application provided invalid PID value, 0x00, for I frame.
[0L] N1LKS>BROCK:(I cmd, n(s)=0, n(r)=1, p=0, pid=0xf0)nodes<0x0d>
N1LKS audio level = 35(19/19)    0002_____
[0.1] N1LKS>BROCK:(I cmd, n(s)=0, n(r)=1, p=0, pid=0xf0)nodes<0x0d>
[0L] N1LKS>BROCK:(RR cmd, n(r)=1, p=1)
BROCK audio level = 97(44/41)   [NONE]   |||||||__
[0.3] BROCK>N1LKS:(DM res, f=1)
Stream 1: AX.25 Protocol Error E: DM received in state 4.
Stream 1: Disconnected from BROCK.
```

wb2osz, 2023-11-03:
Thank you for reporting issues encountered.  There is a lot to digest here.

This is trying to connect with AX.25 version 2.2  (SABME).

Attempting connect to KZ2X-1 ...
[0L] N1LKS>KZ2X-1:(SABME cmd, p=1)

This is a really bad case of audio cross talk.  The transmitted signal is being received.  It is visually confusing but should not hurt.  Direwolf release 1.7 avoids this.

N1LKS audio level = 35(21/19)    334______
[0.1] N1LKS>KZ2X-1:(SABME cmd, p=1)

Connection is successful.     What AX.25 implementation is used at KZ2X-1?    

KZ2X-1 audio level = 103(45/41)    0000001__
[0.3] KZ2X-1>N1LKS:(UA res, f=1)
Stream 14: Connected to KZ2X-1.  (v2.2)

We get  a greeting from the system.

KZ2X-1 audio level = 104(43/41)    0000000__
[0.3] KZ2X-1>N1LKS:(I cmd, n(s)=0, n(r)=0, p=0, pid=0xf0)CONNECTED TO KZ2X TIP #1 ON AX.25 FROM N1LKS<0x0d><0x0d>TYPE 'HOSTS' FOR HOSTS, 'EXIT' TO LOGOUT, 'HELP' FOR HELP<0x0d>TYPE 'CONNECT HOSTNAME' TO CONNECT TO A HOST<0x0d><0x0d>Pad><0x20>

Here we are sending the XID command to negotiate parameters such as packet data size, window size, whether SREJ is available, and so on.   A v 2.2 implementation should be able to process this.

[0L] N1LKS>KZ2X-1:(XID cmd, p=1) Half-Duplex REJ SREJ Multi-SREJ modulo-128 I-Field-Length-Rx=256 Window-Size-Rx=32 Ack-Timer=3000 Retries=10

This makes absolutely no sense at all.   Why are we shutting down the connection?

[0L] N1LKS>KZ2X-1:(DM res, f=1)
[0L] N1LKS>KZ2X-1:(DM res, f=1)

To be continued...

larsks, 2023-11-03:
I'm not sure what kz2x-1 is using (I'll ask), but the Brock digipeater looks like a kantronics of some sort.

wb2osz, 2023-11-06:
I live on Stedman St which crosses the Lowell/Chelmsford boundary.  Do you have any connection to the street name?

wb2osz, 2023-11-06:
Sending two DM packets after receiving an "I" from makes to sense.
After that, it looks pretty normal with the RR and you sending some commands to the BBS.
First, I would really recommend using release 1.7 because some connected mode bugs have been fixed in the past 3 years.
To gain more information, could you add "-d a" to the direwolf command line to list the communication with the client app?
It might also be useful to enable timestamps with "-T %H:%M:%S"

larsks, 2023-11-06:
> First, I would really recommend using release 1.7

I am using release 1.7.

> To gain more information, could you add "-d a" to the direwolf command line to list the communication with the client app?

Sure, I'll do that. I'll try to gather those logs later this evening (or tomorrow if that doesn't work out).

> I live on Stedman St which crosses the Lowell/Chelmsford boundary. Do you have any connection to the street name?

Not as far as I know!

---

To add to the confusion, here's easyterm failing to connect to BROCK:

```
Attempting connect to BROCK ...
[0L] N1LKS>BROCK:(SABME cmd, p=1)
[0.2] BROCK>N1LKS:(DM res, f=1)
BROCK doesn't understand AX.25 v2.2.  Trying v2.0 ...
[0L] N1LKS>BROCK:(SABM cmd, p=1)
[0.2] BROCK>N1LKS:(UA res, f=1)
Stream 10: Connected to BROCK.  (v2.0)
[0.3] BROCK>N1LKS:(I cmd, n(s)=0, n(r)=0, p=0, pid=0xf0)###CONNECTED TO NODE BROCK(W1MV-7) CHANNEL A<0x0d>Welcome to BROCK (W1MV-7) in Brockton, Mass<0x0d>ENTER COMMAND: B,C,J,N, or Help ?<0x0d>
[0L] N1LKS>BROCK:(DM res, f=1)
[0L] N1LKS>BROCK:(DM res, f=1)
[0L] N1LKS>BROCK:(RR res, n(r)=1, f=0)
[0L] N1LKS>BROCK:(I cmd, n(s)=0, n(r)=1, p=0, pid=0xf0)help<0x0d>
[0L] N1LKS>BROCK:(RR cmd, n(r)=1, p=1)
[0.3] BROCK>N1LKS:(DM res, f=1)
Stream 10: Disconnected from BROCK.
```

And then here's the exact same connection working using [qttermtcp](https://www.cantab.net/users/john.wiseman/Documents/QtTermTCP.html):

```
Attempting connect to BROCK ...
[0L] N1LKS-14>BROCK:(SABME cmd, p=1)
[0.3] BROCK>N1LKS-14:(DM res, f=1)
BROCK doesn't understand AX.25 v2.2.  Trying v2.0 ...
[0L] N1LKS-14>BROCK:(SABM cmd, p=1)
[0.3] BROCK>N1LKS-14:(UA res, f=1)
Stream 9: Connected to BROCK.  (v2.0)
[0.3] BROCK>N1LKS-14:(I cmd, n(s)=0, n(r)=0, p=0, pid=0xf0)###CONNECTED TO NODE BROCK(W1MV-7) CHANNEL A<0x0d>Welcome to BROCK (W1MV-7) in Brockton, Mass<0x0d>ENTER COMMAND: B,C,J,N, or Help ?<0x0d>
[0L] N1LKS-14>BROCK:(RR res, n(r)=1, f=0)
[0L] N1LKS-14>BROCK:(I cmd, n(s)=0, n(r)=1, p=0, pid=0xf0)help<0x0d>
[0.3] BROCK>N1LKS-14:(I cmd, n(s)=1, n(r)=1, p=0, pid=0xf0)ABORT           STOP A CONNECTION IN PROGRESS<0x0d>B(ye)           NODE WILL DISCONNECT<0x0d>C(onnect) call  CONNECT TO callsign<0x0d>C call S(tay)   STAY CONNECTED TO NODE WHEN END DISCONNECTS<0x0d>J(heard)        CALLSIGNS WITH DAYSTAMP<0x0d>J S(hort)       HEARD CALLSIGNS ONLY
[0.3] BROCK>N1LKS-14:(I cmd, n(s)=2, n(r)=1, p=0, pid=0xf0)<0x0d>J L(ong)        CALLSIGNS WITH DAYSTAMP AND VIAS<0x0d>N(odes)         HEARD NODE CALLSIGNS WITH DAYSTAMP<0x0d>N S(hort)       NODE CALLSIGNS ONLY<0x0d>N L(ong)        NODE CALLSIGNS WITH DAYSTAMP AND VIAS<0x0d>ENTER COMMAND: B,C,J,N, or Help ?<0x0d>
[0L] N1LKS-14>BROCK:(RR res, n(r)=3, f=0)
[0L] N1LKS-14>BROCK:(I cmd, n(s)=1, n(r)=3, p=0, pid=0xf0)b<0x0d>
[0.3] BROCK>N1LKS-14:(RR res, n(r)=2, f=0)
[0.3] BROCK>N1LKS-14:(DISC cmd, p=1)
Stream 9: Disconnected from BROCK.
```

This seems repeatable (qttermtcp connects successfully when both easyterm and agwterm will not). In all cases, I'm running the clients on the same system and they are connecting to the same Direwolf instance (running on a raspberry pi).

In both cases, the client is configured with a paclen of 128. My `direwolf.conf` looks like:

```
ADEVICE sysdefault:CARD=radio
ARATE 48000
ACHANNELS 1

MYCALL N1LKS
MODEM 1200
CHANNEL 0
PTT /dev/radio RTS

AGWPORT 8000
KISSPORT 8001
```

larsks, 2023-11-07:
The *really* confusing thing is that later in the evening (i.e., now), the connection is suddenly working correctly with both agwterm and easterm.

larsks, 2023-11-11:
Capturing the trace was tricky because the behavior is inconsistent -- at times the connections Just Work, but then something happens and they reliably don't work. I thought that maybe some internal state in Direwolf was getting corrupt, but simply restarting Direwolf does **not** resolve the problem. Connections via the Linux ax.25 stack (e.g. using `axcall`) seem to work reliably, regardless of whether or not AGW applications can successfully connect.

There are two traces here.

First, here's a trace with `-d a` enabled that shows a failure connecting with `agwterm`.

```
<<< Connect, Start an AX.25 Connection from AGWPE client application 0, total length = 36
        portx = 0, datakind = 'C', pid = 0x00
        call_from = "N1LKS", call_to = "BROCK"
        data_len = 0, user_reserved = 20728320, data =
Attempting connect to BROCK ...
[0L] N1LKS>BROCK:(SABME cmd, p=1)
>>> Monitoring Own Information to AGWPE client application 0, total length = 81
        portx = 0, datakind = 'T', pid = 0x00
        call_from = "N1LKS", call_to = "BROCK"
        data_len = 45, user_reserved = 0, data =
  000:  20 31 3a 46 6d 20 4e 31 4c 4b 53 20 54 6f 20 42   1:Fm N1LKS To B
  010:  52 4f 43 4b 20 3c 53 41 42 4d 45 20 50 3d 31 20  ROCK <SABME P=1
  020:  3e 5b 32 30 3a 35 32 3a 30 32 5d 0d 00           >[20:52:02]..
BROCK audio level = 113(24/22)    _|||||___
Audio input level is too high.  Reduce so most stations are around 50.
[0.3] BROCK>N1LKS:(DM res, f=1)
>>> Monitored Supervisory Information to AGWPE client application 0, total length = 78
        portx = 0, datakind = 'S', pid = 0x00
        call_from = "BROCK", call_to = "N1LKS"
        data_len = 42, user_reserved = 0, data =
  000:  20 31 3a 46 6d 20 42 52 4f 43 4b 20 54 6f 20 4e   1:Fm BROCK To N
  010:  31 4c 4b 53 20 3c 44 4d 20 46 3d 31 20 3e 5b 32  1LKS <DM F=1 >[2
  020:  30 3a 35 32 3a 30 33 5d 0d 00                    0:52:03]..
BROCK doesn't understand AX.25 v2.2.  Trying v2.0 ...
[0L] N1LKS>BROCK:(SABM cmd, p=1)
>>> Monitoring Own Information to AGWPE client application 0, total length = 80
        portx = 0, datakind = 'T', pid = 0x00
        call_from = "N1LKS", call_to = "BROCK"
        data_len = 44, user_reserved = 0, data =
  000:  20 31 3a 46 6d 20 4e 31 4c 4b 53 20 54 6f 20 42   1:Fm N1LKS To B
  010:  52 4f 43 4b 20 3c 53 41 42 4d 20 50 3d 31 20 3e  ROCK <SABM P=1 >
  020:  5b 32 30 3a 35 32 3a 30 34 5d 0d 00              [20:52:04]..
BROCK audio level = 111(24/22)    _|||||___
Audio input level is too high.  Reduce so most stations are around 50.
[0.3] BROCK>N1LKS:(UA res, f=1)
>>> Monitored Supervisory Information to AGWPE client application 0, total length = 78
        portx = 0, datakind = 'S', pid = 0x00
        call_from = "BROCK", call_to = "N1LKS"
        data_len = 42, user_reserved = 0, data =
  000:  20 31 3a 46 6d 20 42 52 4f 43 4b 20 54 6f 20 4e   1:Fm BROCK To N
  010:  31 4c 4b 53 20 3c 55 41 20 46 3d 31 20 3e 5b 32  1LKS <UA F=1 >[2
  020:  30 3a 35 32 3a 30 36 5d 0d 00                    0:52:06]..
Stream 0: Connected to BROCK.  (v2.0)
>>> AX.25 Connection Received to AGWPE client application 0, total length = 70
        portx = 0, datakind = 'C', pid = 0x00
        call_from = "BROCK", call_to = "N1LKS"
        data_len = 34, user_reserved = 0, data =
  000:  2a 2a 2a 20 43 4f 4e 4e 45 43 54 45 44 20 57 69  *** CONNECTED Wi
  010:  74 68 20 53 74 61 74 69 6f 6e 20 42 52 4f 43 4b  th Station BROCK
  020:  0d 00                                            ..
BROCK audio level = 101(24/22)    |||||||__
[0.3] BROCK>N1LKS:(I cmd, n(s)=0, n(r)=0, p=0, pid=0xf0)###CONNECTED TO NODE BROCK(W1MV-7) CHANNEL A<0x0d>Welcome to BROCK (W1MV-7) in Brockton, Mass<0x0d>ENTER C>
>>> **INVALID** to AGWPE client application 0, total length = 222
        portx = 0, datakind = 'I', pid = 0x00
        call_from = "BROCK", call_to = "N1LKS"
        data_len = 186, user_reserved = 0, data =
  000:  20 31 3a 46 6d 20 42 52 4f 43 4b 20 54 6f 20 4e   1:Fm BROCK To N
  010:  31 4c 4b 53 20 3c 49 20 53 30 20 52 30 20 70 69  1LKS <I S0 R0 pi
  020:  64 3d 46 30 20 4c 65 6e 3d 31 32 33 20 50 3d 30  d=F0 Len=123 P=0
  030:  20 3e 5b 32 30 3a 35 32 3a 30 37 5d 0d 23 23 23   >[20:52:07].###
  040:  43 4f 4e 4e 45 43 54 45 44 20 54 4f 20 4e 4f 44  CONNECTED TO NOD
  050:  45 20 42 52 4f 43 4b 28 57 31 4d 56 2d 37 29 20  E BROCK(W1MV-7)
  060:  43 48 41 4e 4e 45 4c 20 41 0d 57 65 6c 63 6f 6d  CHANNEL A.Welcom
[... 285 lines cut ...]
  0e0:  0d                                               .
[0L] N1LKS-14>BROCK:(RR res, n(r)=3, f=0)
>>> Monitoring Own Information to AGWPE client application 1, total length = 84
        portx = 0, datakind = 'T', pid = 0x00
        call_from = "N1LKS-14", call_to = "BROCK"
        data_len = 48, user_reserved = 0, data =
  000:  20 31 3a 46 6d 20 4e 31 4c 4b 53 2d 31 34 20 54   1:Fm N1LKS-14 T
  010:  6f 20 42 52 4f 43 4b 20 3c 52 52 20 52 33 20 46  o BROCK <RR R3 F
  020:  3d 30 20 3e 5b 32 30 3a 35 37 3a 35 35 5d 0d 00  =0 >[20:57:55]..
<<< Send Connected Data from AGWPE client application 1, total length = 38
        portx = 0, datakind = 'D', pid = 0xf0
        call_from = "N1LKS-14", call_to = "BROCK"
        data_len = 2, user_reserved = 0, data =
  000:  62 0d                                            b.
[0L] N1LKS-14>BROCK:(I cmd, n(s)=1, n(r)=3, p=0, pid=0xf0)b<0x0d>
>>> Monitoring Own Information to AGWPE client application 1, total length = 102
        portx = 0, datakind = 'T', pid = 0x00
        call_from = "N1LKS-14", call_to = "BROCK"
        data_len = 66, user_reserved = 0, data =
  000:  20 31 3a 46 6d 20 4e 31 4c 4b 53 2d 31 34 20 54   1:Fm N1LKS-14 T
  010:  6f 20 42 52 4f 43 4b 20 3c 49 20 53 31 20 52 33  o BROCK <I S1 R3
  020:  20 70 69 64 3d 46 30 20 4c 65 6e 3d 32 20 50 3d   pid=F0 Len=2 P=
  030:  30 20 3e 5b 32 30 3a 35 37 3a 35 39 5d 0d 62 0d  0 >[20:57:59].b.
  040:  0d 00                                            ..
```

va2bbw, 2025-08-30:
I wanted to add my own experience to this thread, in case it is related. See my original issue here I had originally posted to the Paracon project: https://github.com/mfncooper/paracon/issues/26 
I'd love to add some more information to this, I would just need to know what you need that would be valuable to you to figure this out.
Just to reiterate my issue so you don't have to check the link, here's what happens:

1) I'm using Direwolf and Paracon. 
2) With Paracon, I try to connect to a BBS, and it shows that I do so succesfully. 
3) As soon as I send a command to the BBS, or any other string of text, I get disconnected.

Right now that's what I've got. Paracon author pointed me here as he suspects it may possibly be related. Would love to get this resolved so I can use Paracon. So far this behaviour happens every single time.

jmkristian, 2025-08-31:
Perhaps it would be informative to try a different AGWPE client (with the same Direwolf and BBS). I suggest `converse` from [agwpe-tools](https://github.com/jmkristian/agwpe-tools).

mfncooper, 2025-08-31:
The issue has been seen with agwterm, QtTermTCP and Paracon, but sure, there's no harm in trying one more. :-)