wb2osz/direwolf #504: Successful-APRS-IGate-Operation.pdf not available

oe5reo, 2023-11-26:
Just noticed that the PDF "Successful-APRS-IGate-Operation" seems to be missing in the /doc directory.

dranch, 2023-11-26:
Hmmm.. works for me.  That doc is present in the master repo: https://github.com/wb2osz/direwolf/blob/master/doc/Successful-APRS-IGate-Operation.pdf and in the dev branch at https://github.com/wb2osz/direwolf/blob/dev/doc/Successful-APRS-IGate-Operation.pdf .  Maybe when you say "/doc" directory, do you mean /usr/share/doc/direwolf ?  That also works for me when Direwolf is installed as .deb package:  $ file /usr/share/doc/direwolf/Successful-APRS-IGate-Operation.pdf
/usr/share/doc/direwolf/Successful-APRS-IGate-Operation.pdf: PDF document, version 1.5

wb2osz, 2023-11-26:
Could you be more specific?
Which doc directory?

oe5reo, 2023-11-27:
Sorry, this was my fault! The document I was looking for can now be found on https://github.com/wb2osz/direwolf-doc instead of https://github.com/wb2osz/direwolf

I will soon update my portable iGate OE5REO-6 from V1.6 to V1.7 and wanted to have a quick look at all the available documents to prevent making a lot of mistakes (I am far from being an expert ...).

73 Martin
OE5REO