wb2osz/direwolf #505: AX.25/IL2P Switch

flux7782, 2023-12-06:
Would it be possible, or does direwolf today support the ability to be a frame switch ? So unlike a digipeater, it would receive a frame on one RF interface and pass that frame on to another RF interface based on a lookup table. Is this something that requires direwolf to be attached to an external program today ? Thanks in advance

wb2osz, 2023-12-06:
Dire Wolf already has a REGENERATE option which causes an incoming frame to be resent without modification. 
It is possible to define filters so packets are forwarded to different radio channels based on properties.
Please be more specific about what you mean by lookup table.

flux7782, 2023-12-06:
Lookup Table - Not unlike an ethernet mac table, so that you could use direwolf to build a bridge/switch that would only forward frames to ports that it had a destination for

wb2osz, 2023-12-07:
https://groups.io/g/direwolf/