wb2osz/direwolf #513: BLE KISS Support

8cH9azbsFifZ, 2024-02-05:
It would be great to have BLE KISS support in this software modem: https://github.com/hessu/aprs-specs/blob/master/BLE-KISS-API.md.

dranch, 2024-02-07:
As an interesting side point, check out this ESP32 project:  https://github.com/islandmagic/bb-link/

Some devices, like the Kenwood TH-D74 radio only support Bluetooth Classic serial profile. iOS devices only support Bluetooth Low Energy (BLE). They are not compatible and as such, you can't pair those devices together. This code provides a way to create an adapter that can interface a device that exposes a serial profile over Bluetooth Classic, to an iOS device via BLE. Its main purpose is to enable iOS app that supports AX.25 packet like RadioMail or APRS.fi to use the TNC built in the radio as a modem.

wb2osz, 2024-02-07:
See  https://github.com/wb2osz/direwolf/blob/master/doc/Bluetooth-KISS-TNC.pdf  
direwolf does not implement bluetooth.  
There is operating system configuration which maps bluetooth to a virtual serial port.
direwolf then uses this virtual serial port.

Maybe BLE would be the same where an OS configuration would allow it to appear as a serial port to applications.
Good suggestion, but I don't understand bluetooth and don't have an iOS device for experimenting.
Perhaps someone, in the community, who knows more about Bluetooth, can tweak the configuration to use BLE rather than classic and tell me what to put in the documentation.