wb2osz/direwolf #522: Direwolf 1.7 igate no TX

ve7ltn, 2024-03-25:
I'm running Direwolf 1.7 on a Pi Zero 2W. In Direwolf, log entries in pink are supposed to indicate what is being transmitted out of the radio.  The radio isn't Tx'ing. The radio will Rx and send packets to APRS-IS, but packets coming back don't Tx, even though direwolf says it has TX'd. 
DIGIPEAT 0 0 ^WIDE[3-7]-[1-7]$|^TEST$ ^WIDE[12]-[12]$ TRACE is enabled
IGSERVER noam.aprs2.net
IGLOGIN WB2OSZ-5 123456 <-- with my call and passcode
IGTXVIA 0 WIDE1-1 is enabled
IGTXLIMIT 6 10 is enabled

Any suggestions?  cable? radio? I'm using the digirig and cables for Baofeng/Kenwood on a Retevis RT85 (should be same pinouts as Baefeng/Kenwood).  Tested the volume levels as well.  Any assistance is greatly appreciated.  Thanks. 73

dranch, 2024-03-25:
Github "issues" are not the right way to get support unless you think you have a real bug (this specific scenario is unlikely).  Please join and repost this issue to direwolf@groups.io.  In that email, please include your direwolf.conf file, show the direwolf startup output, and please mention if Direwolf can successfully TX on RF for say basic beacons, etc.

Please close this github issue.

--David
KI6ZHD

wb2osz, 2024-03-26:
This is for reporting defects and making enhancement suggestions.
Please use  https://groups.io/g/direwolf/  to ask for help.
Be sure to include your direwolf configuration file and everything you see on the screen to provide more clues about your TX problem.

73,
John WB2OSZ

closed 2024-03-26 by wb2osz