wb2osz/direwolf #523: DWAIT Not working

na7q, 2024-03-31:
Latest dev branch, the DWAIT no longer seems to work.

This is tested on Armbian 12 (Debian 12). No matter the settings, it does not seem to work.

na7q, 2024-03-31:
Did a full reinstall of 1.7a, worked fine, went back to latest dev, and it started working. Unsure if related to Debian 12, or Direwolf. Closing for now.

closed 2024-03-31 by na7q

wb2osz, 2024-04-01:
What makes you think you need the DWAIT option?

na7q, 2024-04-03:
I use it to better time digipeating between my digis and some distant digis. Otherwise we just clash most of the time. If we clash, we don't have a well coordinated working network. It also helps that receive transmit turn around time.
Having this option and running it for many years now has proven to be very effective. A must have at every deployment I've made.

I'm still not sure why it initially didn't work.