wb2osz/direwolf #524: Kiss port 8001 issue

SpudGunMan, 2024-04-12:
Direwolf is always binding 8001

Config is minimal, sound and defining kiss and agw ports as custom port numbers. 

When 1.8a march 9 build is ran, it will bind the 8001 and the config's defined kiss simultaneous unable to stop the bind to port 8001 

Mint-Jessie

wb2osz, 2024-04-12:
Could you be more specific, please?

Exactly what is in the configuration file?

What does direwolf display?

SpudGunMan, 2024-04-13:
dire2.conf
```
ADEVICE  null null

MYCALL K7MHI

AGWPORT 8100
KISSPORT 8101
```

spud@k7mhi-dev:~$ direwolf -c dire2.conf 
```
Dire Wolf DEVELOPMENT version 1.8 A (Mar  9 2024)
Includes optional support for:  gpsd hamlib cm108-ptt libgpiod

Reading config file dire2.conf
Audio device for both receive and transmit: null  (channel 0)
Channel 0: 1200 baud, AFSK 1200 & 2200 Hz, A+, 44100 sample rate.

Ready to accept AGW client application 0 on port 8100 ...
Ready to accept KISS TCP client application 0 on port 8001 ...
Ready to accept KISS TCP client application 0 on port 8101 ...
```

dranch, 2024-04-13:
It might be worth mentioning that this issue has been reported before:

  https://github.com/wb2osz/direwolf/issues/342

SpudGunMan, 2024-04-13:
Sorry didnt search issues well enough, closing this one

closed 2024-04-13 by SpudGunMan