wb2osz/direwolf #531: Missing "*" for set h bit after more than one digipeater callsign

DL7NDR, 2024-05-26:
Satellite [EOS-7](https://db.satnogs.org/satellite/55562) (A55BTN) answers the packet 'Fm DL7NDR to CQ via ARISS' by 'DL7NDR>CQ,A55BTN*,ARISS*'.
EOS-7 always inserts his own callsign (A55BTN) and sets the h bit on both callsign.
This should be shown by a  *  after ARISS -AND- after A55BTN as it is using SoundModem.
The current behaviour of Direwolf is to show the  *  only after ARISS.

[image]

komacke, 2024-09-12:
I see DL7NDR created PR 530. I've been using that PR with paracon (https://github.com/mfncooper/paracon) in multiple unproto-based nets and it's working as expected. It would be great to get this update into a release.

wb2osz, 2024-09-13:
Thank you for your feedback.
You are confusing two different things.

The first image has the standard TNC-2 monitoring format, where only the LAST "used" (h bit set) address is marked with *.
See "Understanding APRS Packets" in https://how.aprs.works .

UZ7HO SoundModem has its own, very different, format, not the standard TNC-2 Monitoring format.
It has * for all used addresses.

Please see most recent comment in Issue [530](https://github.com/wb2osz/direwolf/pull/530).

closed 2024-09-13 by wb2osz