wb2osz/direwolf #532: Binding failed with error 10013 - Cant bind to ANY ports

W1BTR, 2024-05-30:
Direwolf suddenly is failing to bind to any ports.

[image]

```
Bind failed with error: 10013
Some other application is probably already using port 8000.
Try using a different port number with AGWPORT in the configuration file.
Bind failed with error: 10013
Some other application is probably already using port 8001.
Try using a different port number with KISSPORT in the configuration file.
```

I was able to verify that no programs are using these ports. I also tried setting it to a variety of other ports I know are available, and they all fail with error 10013.

[image]

dranch, 2024-05-30:
Hmmm.. we need more details here before we can help.  It looks like you're running Windows but WHICH version?  Which version of Direwolf?  Googling around a bit, it could be security software installed on your machine that's blocking Direwolf from opening the socket ( https://forum.beammp.com/t/core-bind-failed-with-error-10013/102578 ).

wb2osz, 2024-05-30:
Interesting.  I tried the netstat command as shown and it behaves as expected.
I have Windows 10 here,  11 might be fussier.
I don't have any good ideas other than what David suggested.
Can you try running with admin privilege?

73,
John WB2OSZ

W1BTR, 2024-06-03:
It is Windows 11, and I tried running as admin, no change. Restarting didnt fix it. However, a Windows update did fix it, so maybe Microsoft broke it temporarily with an update?

W1BTR, 2024-06-03:
It is Windows 11, and I tried running as admin, no change. Restarting didnt fix it. However, a Windows update did fix it, so maybe Microsoft broke it temporarily with an update?

Either way, cant seem to get it to break now.

closed 2024-06-03 by W1BTR