wb2osz/direwolf #533: Direwolf does not support VARA compatibility

DrFrankReade, 2024-06-06:
Well, no kidding, of course it doesn't. 

But given how well Direwolf works and how practical and well thought out the interface is, wouldn't it be nice if the community could come together and fork a VARA-compatible version, in order to secure what is a popular modulation scheme that is unfortunately in danger of unexpected extinction at any time due to the fact that it's much like the last northern white rhino, and living out its on a Wordpress blog and it only runs on Windows? Naturally, this venture might irritate, but maybe not as irritating as a dedicated Windows box. And it may even save Vara, as the scheme is so poorly documented, there are questions about whether it's even kosher for (at least US) operators to use - After all, I can't at all be certain of what data is actually going over the air, nor can anybody else. Which is the whole point of not being allowed to use encryption on amateur bands.  So while this isn't an issue at all, and scold me if you must, I do feel that this is a hornets nest that would be worth kicking. So here I go. 

https://github.com/pengowray/VaraHuffmanNet/issues/1

wb2osz, 2024-06-06:
Rather than incorporating VARA capability, it would make more sense to use the TCP KISS interface to another application.
https://mega.nz/file/eCxBFZzA#F6mfrEtQ0UCh7dxxKinwgU0livyB_87QI2delWbYn_U 

Would you be willing to perform the testing if such an interface was added?

73,
John WB2OSZ

srsampson, 2024-06-06:
This issue is a non-starter. Proprietary _undocumented_ protocols should not be encouraged. Proprietary _documented_ protocols are desirable, even though non-proprietary systems would attract the most developers. For example, the IL2P additions are making the protocol available to desktop users, rather than limited to firmware appliances.

F4FXL, 2024-06-11:
Generally speaking, if Direwolf could handle external TNCs over kiss this would open up a brand new world.  Interfacing with LoRa, Robust Packet .....

wb2osz, 2024-08-04:
According to https://pslara.org/docs/314.D.pdf , page 22, VARA FM has the ability to use KISS over TCP.

The direwolf development branch ("dev") now has the capability to map a virtual channel to an external TNC over TCP KISS.

The example given is for LoRa, but VARA FM, or any other modem supporting TCP KISS should work.

https://github.com/wb2osz/direwolf-doc/blob/main/APRS-LoRa-VHF-APRS-Bridge.pdf

Give it a try and tell us how it worked out.

73,
John WB2OSZ

F4FXL, 2025-09-16:
Hi John, I've sent you an email in this regard.