wb2osz/direwolf #539: What is the coordinate format that is used in the configuration file?

mikeym88, 2024-07-21:
I'm taking a look at the beaconing properties in the configuration file here: https://github.com/wb2osz/direwolf/blob/master/conf/generic.conf#L420-L429. The coordinate format is one that I do not recognize. What is the format? Is the first number the degrees and the second number is the minutes and seconds combined into 1 decimal number?

mikeym88, 2024-07-21:
Confirmed: the second number is the [minutes and seconds combined into a decimal number](https://www.maptools.com/tutorials/lat_lon/formats#:~:text=and%20Decimal%20Minutes-,DDD%C2%B0%20MM.MMM%27,-32%C2%B0%2018.385%27%20N).

closed 2024-07-21 by mikeym88

wb2osz, 2024-07-23:
The format is degrees ^ minutes and fractional minutes.
The GPS NMEA and APRS positions/objects also use degrees and minutes and fractional minutes without a separator after degrees.
This is explained in the User Guide.

73,
John WB2OSZ