wb2osz/direwolf #542: TOCALL ADD:

richonguzman, 2024-08-16:


dranch, 2024-08-16:
Both are present in https://raw.githubusercontent.com/aprsorg/aprs-deviceid/main/generated/tocalls.pretty.json

richonguzman, 2024-08-16:
ok

this "issue" started as I did not find it here:
https://github.com/wb2osz/direwolf/blob/69407ccf84c443d7da93ef9a63a1ebe7eaf88fce/data/tocalls.txt#L12

but if it loads the data from "tocall.json" file my issue was wrongly started :)

wb2osz, 2024-08-19:
Earlier versions, including 1.7 (current master branch), used the tocalls.txt file which has not been maintained for a couple years.

The current authority is https://github.com/aprsorg/aprs-deviceid/blob/main/tocalls.yaml  .

The development ("dev") branch, which will become release 1.8, uses tocalls.yaml .

$ grep APL data/tocalls.yaml
...
 - tocall: APLRG?
 - tocall: APLRT?
...

73,
John WB2OSZ

closed 2024-08-19 by wb2osz

richonguzman, 2024-08-19:
got it ;)

richonguzman, 2024-08-25:
hi, i've messaged you on Discord if this does not work can I write you mail?

richonguzman at gmail