wb2osz/direwolf #543: AGWPE interface does not mark received digipeater

komacke, 2024-08-29:
Typically a digipeated packet has the received digipeater marked with an asterisk (*) after the respective digipeater. We see this in direwolf's logs and on the direwolf kiss interface. However we don't see it via direwolf's AGWPE interface.

As discussed here (https://github.com/mfncooper/paracon/discussions/8) it seems the use of * is not defined for AGWPE. Some applications include it (ldsped) and some applications don't (direwolf). Without it, there's really no way to tell what the transmitting station was.

Could I request that direwolf include the * of the received digipeater on the AGWPE interface?

Thanks

komacke, 2024-09-02:
Check it out. PR 530 has this change. I'll give this a try with paracon.

wb2osz, 2024-10-19:
Duplicate of [issue 530](https://github.com/wb2osz/direwolf/pull/530) which has already been completed.

closed 2024-10-19 by wb2osz