wb2osz/direwolf #544: Audio Input Timeout Timer is Too Low

jeromegoddard2, 2024-09-06:
Thanks so much for all the work you've done with direwolf--it's a great program.  My issue is that there is an audio timeout timer for input that is set by default as 25ms.  Several members of our local ham radio club have been using direwolf as a part of an iGATE/digipeater system and have encountered direwolf crashing unexpectedly.  After many hours of debugging, we found that our rigs have small gaps in the input audio stream of anywhere from 5ms up to 500ms or so.  These are gaps are unpredictable and distributed sometimes as quickly as minutes apart and sometimes as far apart as days.  

   Thus, I would request that you move this timer setting to a higher default value (say, 1000ms at least) or, even better, allow this timer value to be set in the configuration file.  I edited the source myself and hard coded the timer value to be 10000ms and have not had any more issues.  However, I would rather not have to make this edit every time y'all come up with new features!

Respectfully,
Jerome
W4IJG

dranch, 2024-09-06:
Could you paste in a patch or better yet, create a Git merge request that shows your proposed change?

jeromegoddard2, 2024-09-06:
Thanks for the quick response.

Here is what I changed:

In audio_win.c
Around line 789: In case AUDIO_IN_TYPE_SOUNDCARD: I changed "int timeout = 25;" to "int timeout = 10000;"  

25ms is just too small for our systems.

This small change works well for me.  However, as I suggested, it is probably better to let this be a user defined variable from the config file.  I will let you decide.

Please let me know if you need any more information.

Thanks,

Jerome
W4IJG

dranch, 2024-09-06:
Can you tell us about the computer hardware (CPU/RAM/etc), operating system and version, sound device (make/model, is it a real vendor offering or clone), and how the sound device is connected to the computer as well and if it's a direct connection or it goes through say passive/passive USB hubs?  All this detail matters as 25ms is a LONG time and while this change might improve reliability with Direwolf, this might be masking a far larger issue with your specific setup.

wb2osz, 2024-09-06:
I'm not against making the timeout time longer.   Better to keep retrying longer instead of giving up too soon.  Like David, I'm wondering what sort of situation produces this issue.  I've never seen it, even with a Raspberry Pi model 1 while doing other things on it.

Oh, I see it is on Windows.  "audio_win.c"  What sort of audio interface is it using?  Is there a USB hub between the computer and a USB audio adapter?  Is the computer doing something else very demanding at the time?

John

wb2osz, 2024-10-19:
25 is not milliseconds.
25 is the maximum number of times it will wait for ONE_BUF_TIME (10 milliseconds) so the total is a quarter of a second.
I suppose I could increase 25 to 200 which would be 2 seconds.
Something is SERIOUSLY wrong if there is a 2 second gap in the input audio samples being delivered.

You observed:  "I listened to the audio input from the digirig through the Windows sound control panel and could clearly hear the normal radio input go silent sporadically at random intervals..."

Can you eliminate the USB hub between the USB audio device and the computer?
I've seen USB hubs causing audio problems before.
Does it receive reliably if the transmitter is disabled or turned to low power?
Often transmitter RF will get into the USB device and cause it to malfunction.
Did you try adding ferrite beads around all your cables?

If something is really messed up, I think it is better to abort the program and have it restart automatically rather than sitting there hung.

73,
John WB2OSZ