wb2osz/direwolf #548: Monitor PTT/DCD status

craigerl, 2024-10-07:
It's impossible to monitor a **gpio output pin** status with the new/robust libgpiod and gpiozero API's, and they have no intention of implementing this feature (which existed for years in the /sys/class/gpio filesystem that is now gone).

Direwatch (github.com/craigerl/direwatch) used to watch the DCD and PTT pins (via the missing /sys/class/gpio interface) to let the operator know when the device is receiving and transmitting, drawing the corresponding red and green circles on a little display.

Is there another way to get a near-real-time ptt/dcd status from direwolf?  named pipe? port?  other?  via python?

curiously,
-craig

dranch, 2024-10-07:
There is the "-d o" option to see the status of PTT and DCD via STDOUT.  I suppose you could also wire your PTT GPIO pin to a 10k resistor and over to a different GPIO pin in read-mode and poll that.

craigerl, 2024-10-07:
Thanks David, that may be the trick then, for those playing along at home, send PTT and DCD status to direwolf.ptt.log and send the unaltered log to direwolf.log .   It might be a bit laggy, but it's better than the new GPIO.

` direwolf -d t -d o -p -q d -t 0 -c /run/direwolf.tnc.conf 2>&1 | tee >(grep --line-buffered -E '(DCD |PTT )' > direwolf.ptt.log) | ( grep --line-buffered -E -v '(PTT|DCD)' > direwolf.log )`

```
PTT 0 = 1
PTT 0 = 0
DCD 0 = 1
DCD 0 = 0
PTT 0 = 1
PTT 0 = 0
DCD 0 = 1
DCD 0 = 0
DCD 0 = 1
DCD 0 = 0
```

thanks,
-craig