wb2osz/direwolf #552: Use Direwolf for Raw Packets not AX.25 Frames

buddylindsey, 2024-10-22:
I am trying to write some software and I want to send raw data across RF (mostly experimentation) at this point.

However, I am lost in configuration. I would like to keep it simple and just write the output to a `.wav` file and then read from a `.wav` file while testing my code.

It seems like I would want to use KISS serial tnc, but I am a little lost in the configuration. 

Is there a "simple" way to do non ax.25 data sending and receiving to a serial port for my app to connect to?

ShawnStoddard, 2024-10-22:
Are trying to send raw audio? Can you provide some more details?

buddylindsey, 2024-10-22:
I want to see about messing with writing my own protocol as a learning exercise. Or re-implement ax.25 on my own, or at least part of it. I am trying to better understand lower-level packet radio. However, I feel overwhelmed on the actual generating audio and reading audio portion I was hoping I could use direwolf to handle modulation and demodulation of data.

What I would like to do is send data to direwolf via a serial port and it modulate it to a .wav file. 

I then want to have direwolf read that .wav file demodulate it and push the data to a serial port for my app to read.

I don't want direwolf to do any transformations or try to read the data as ax.25 frames. I don't have a problem wrapping my data with the KISS protocol. Just don't want it to do anything beyond that.

Once i work out what I am trying to do in code I want to then try to send it to direwolf to have direwolf send it to my digirig for transmission in a real scenario.

dranch, 2024-10-22:
Direwolf Github issues aren't the place to ask questions or get support.  That's what the Direwolf group.io list at https://groups.io/g/direwolf is for.  I recommend to create a thread there for getting a LOT more exposure to your question but you also might want to check out Kamal's minimodem - http://www.whence.com/minimodem/ as one idea.  Please close this Github issue.

buddylindsey, 2024-10-22:
Thanks, I missed the groups.io group. I'll go there and will also checkout minimodem.