wb2osz/direwolf #554: Have the startup of Direwolf v1.8 show the complete path for the configuration file used

dranch, 2024-10-31:
See https://groups.io/g/direwolf/message/10002 - could be a regression from 1.7

dranch, 2024-11-01:
another comment from Robert as a reminder to align the documentation too:

Funny, but that is what both the V1.6 (pg 40) and V1.7 (p43) User Manual shows:
     Dire Wolf version ...
     Reading config file /Users/<Home_Dir>/direwolf.conf
But later in the manual, the path section is removed.
-------
Robert Giuliano
KB8RCO

mfncooper, 2025-07-28:
To clarify what is actually printed in the "Reading config file" line (and hasn't changed for 10 years):

* If the config file is specified using '-c', that path will be printed.
* If the config file is found in the current directory, only 'direwolf.conf' will be printed.
* On Linux or Mac, if the config file is found in the home directory, the full path and filename will be printed.

wb2osz, 2025-09-16:
The full path is now displayed to reduce confusion.