wb2osz/direwolf #577: deviceid test fails for APAX

doismellburning, 2025-07-16:
Running the tests in `deviceid.c` with [`gcc -DTEST deviceid.c && ./a.out`](https://github.com/wb2osz/direwolf/blob/0a28e0012ca3bcf76aa90f0a92264810ec4b55c3/src/deviceid.c#L33C1-L33C75) I get a test failure with the following stanza:

https://github.com/wb2osz/direwolf/blob/0a28e0012ca3bcf76aa90f0a92264810ec4b55c3/src/deviceid.c#L153-L156

```
a.out: src/deviceid.c:156: main: Assertion `strcmp(device, "AFilterX") == 0' failed`
```

`device` appears to end up as `"UNKNOWN vendor/modelAFilterX"` instead of the expected `"AFilterX"`

Digging a little, this looks like an artifact of b26c5a4d7d4b2fd19160a1ff20cece13b2c18b13 where the error message was improved, but the test wasn't updated.

I think the best things to do here are probably:

1. Fix `deviceid_decode_dest` to put a space between the error message and the model
2. Fix the test's expected value
3. Automatically run the test as part of the build GitHub Actions

I'll follow up with a PR for 1+2, but 3 might take me longer as I need to grapple with CMake...

doismellburning, 2025-07-16:
Actually I misread that sorry - the vendor is unknown, the model is known, but the error is generic. So I think https://github.com/wb2osz/direwolf/commit/b26c5a4d7d4b2fd19160a1ff20cece13b2c18b13 has a slight bug in the implementation, and the test case is what is intended

wb2osz, 2025-10-02:
Pretty much like Martin suggested.
In dev branch of course.

closed 2025-10-02 by wb2osz