wb2osz/direwolf #585: Non-UI frames being sent as "monitoring own" via AGWPE

mfncooper, 2025-07-28:
From the AGWPE spec:

> **Monitoring Own Information (‘T’ Frames)**
> 
> All information _sent unproto by the application thru the ‘M’ frame_ is returned by AGWPE once transmitted as a ‘T’ frame, this frame could be used for confirmation purposes.

(Italics mine.)

Direwolf is actually sending _all_ sent frame types via 'T' frames, rather than only unproto frames.

mfncooper, 2025-07-30:
I'm aware of different versions of the spec, though I haven't seen the one you refer to dated July 29, 2025. I'd be interested to know where you found that, since even Google couldn't find it for me. The one I generally use is the same spec, but an older version, with a date of December 12, 2000.

The 2000 version includes the text that you cite, but it's under `Send Data in “raw” AX.25 format (‘K’ frame)`, and is saying that a 'K' frame should only be used to send unproto info. As such the "it" in "it could be used to send both connected and unconnected information" is a 'K' frame. The later "When connected information is sent using this frame [meaning a 'K' frame] the application will not receive monitor frames (T frames)" is consistent with the documentation for 'T' frames saying that they are sent only for unproto info.

I realise that the text you cited may be in a different place in the version of the spec you're looking at, but that's my interpretation based on the earlier version I have.

wb2osz, 2025-08-03:
Try the "dev" branch and see if it behaves like you want.

mfncooper, 2025-08-04:
Yup, that's what I would expect to see. Thanks!