wb2osz/direwolf #590: Debian 13 Trixie gpiod API change

wb2osz, 2025-08-16:
Thanks to Dave and Martin for pointing this out.

direwolf fails to build on Debian 13 Trixie because one of the gpiod API functions has been removed.

Here's the commit to libgpiod that removes the 'gpiod_chip_open_by_name' function, with the explanation for why it was removed:

https://github.com/brgl/libgpiod/commit/eb690e46efd09cd9a758ed1b19f99601f7b5de8a

wb2osz, 2025-08-16:
This is the only remaining open function:

/**
 * @brief Open a chip by path.
 * @param path Path to the gpiochip device file.
 * @return GPIO chip object or NULL if an error occurred. The returned object
 *         must be closed by the caller using ::gpiod_chip_close.
 */
struct gpiod_chip *gpiod_chip_open(const char *path);

leventelist, 2025-08-24:
I think it is because the API changed from 1.6.x to 2.x. I am now doing the upgrade in FLDIGI. If I have time, I try to fix this in DW.

wb2osz, 2025-08-24:
I think I have a fix but then I broke something else because I had some partially completed work, for something else, in one of the modified files.  Most of the open functions disappeared so instead of ...open_by_name ("gpiochip0") we need to use the full device path like ...open("/dev/gpiochip0").

leventelist, 2025-08-24:
@wb2osz Can you send me your patch? I'm not sure what has changed.

wb2osz, 2025-08-24:
Might be fixed by commit ade6d0e .   

Runs OK on RPi 5 bookworm but haven't installed Trixie yet.

wb2osz, 2025-08-24:
Upgrading my RPi 5 from bookworm to trixie in-place (i,e. not a fresh install) to (hopefully) preserve other stuff I'm working on.
Using instructions from https://forums.raspberrypi.com/viewtopic.php?t=389477&hilit=updating+trixie

wb2osz, 2025-08-24:
Good news:  I did not break it on bookworm.
Bad news:  Looks like they removed another function I'm using.

/home/pi/direwolf/src/ptt.c:669:16: error: implicit declaration of function ‘gpiod_chip_get_line’; did you mean ‘gpiod_chip_get_info’? [-Wimplicit-function-declaration]
  669 |         line = gpiod_chip_get_line(chip, line_number);
      |                ^~~~~~~~~~~~~~~~~~~
      |                gpiod_chip_get_info

Closer to the goal but not there yet.

dranch, 2025-08-24:
I installed the real Debian 13 / Trixie on a x86-64 VM but treated the VM as if it was a Raspberry Pi by installing all the GPIO and other libraries used by Direwolf.

wb2osz, 2025-08-24:
It's a lot more complicated than I first thought.  "The v2.x API is a complete redesign."  

Common Migration Pitfalls
1. Assuming 1:1 API Mapping
- Reality: The v2.x API is a complete redesign. Functions like gpiod_chip_get_line() are gone, and there's no direct replacement — you now use a request-based model.
- Fix: Refactor your logic to use gpiod_chip_request_lines() and manage lines via gpiod_line_request.
2. Not Updating Line Access Logic
- Old: You accessed a line, then set direction and value.
- New: You define direction and settings before requesting the line.
- Fix: Use gpiod_line_settings and gpiod_request_config to configure everything upfront.
3. Confusion Over Multi-Line Requests
- Reality: v2.x encourages multi-line requests, even if you're only using one line.
- Fix: Always pass an array of offsets, even if it's just one.
4. Missing Consumer Labels
- Impact: If you don’t set a consumer label, debugging becomes harder.
- Fix: Always set .consumer = "your-app-name" in gpiod_request_config.
5. Incorrect Value Access
- Old: gpiod_line_get_value(line)
- New: gpiod_line_request_get_value(request, offset)
- Fix: Use the request object to get/set values.
6. Not Handling Events Properly
- Reality: Event handling is more structured now, but also more complex.
- Fix: Use gpiod_line_event_buffer and related APIs for edge detection.
7. Header and ABI Confusion
- Reality: Some distros package v2.x as libgpiod2 or libgpiod3 due to ABI changes.
- Fix: Double-check your build system and headers to ensure you're linking against the correct version.

leventelist, 2025-08-25:
So far, I have this. https://github.com/leventelist/fldigi_dev/blob/gpiod_integration/src/gpio_common/gpio_common.c

Pandapip1, 2025-08-26:
Might be worth opening an issue upstream to have the config.h included in distributions?

leventelist, 2025-08-30:
I have some working prototype for FLDIGI. You might want to take a look.

https://github.com/leventelist/fldigi_dev/compare/pu/df...gpiod_integration

leventelist, 2025-08-30:
I started working on this. https://github.com/leventelist/direwolf/tree/gpiod_support

wb2osz, 2025-09-02:
A proposed fix has been added to dev branch.

This has been tested with:
RPi 3 bookworm - libgpiod v1
RPi 5 trixie - libgpiod v2
Windows 11 - no gpiod

Your feedback is appreciated.

dranch, 2025-09-03:
Builds 100% on Debian Trixie x86_64 VM with libgpiod 2.2.1
Builds 100% on Ubuntu 24.04 x86_64 bare metal with libgpiod2t64 1.6.3
Builds 100% on Raspberry Pi OS Bookworm ARM64 on Rpi3B+ with libgpiod 1.6.3

closed 2025-09-15 by wb2osz