wb2osz/direwolf #591: Direwolf 1.7.0-2260df1i686 blocked due to virus

Bigjohn4, 2025-08-26:
I downloaded and unzipped Direwolf 1.7.0-2260df1i686. When I first ran dierewolf.exe, Norton AV stopped it and said it had IDP.Generic virus and would not let it run.

Any fix or workaround?

Thanks, John kb5ggo

dranch, 2025-08-26:
This is most likely a false positive - please search the Internet with "Norton AV stopped a process saying it had IDP.Generic virus" to get more details and how to disable this check for the Direwolf binary.

Bigjohn4, 2025-08-27:
Thanks. I'll look into it. I did send it to Norton.

JM

wb2osz, 2025-08-27:
I've seen a few false positives with Norton. with very innocent little applications that I compiled from source.
My understanding is that is impossible to keep up with all new malware variations, so they use some sort of heuristic to detect anything that looks out of the ordinary.
direwolf is built with a version of gcc, rather than the Microsoft tools, so that that might look a little unusual.

If you used the version from https://github.com/wb2osz/direwolf/releases it is approaching 2 years old and is used by many people.  It should be perfectly safe.

If you downloaded it from somewhere else, then all bets are off.

Norton has an option somewhere that allows you to set scanning exceptions for files or folders.

Thank you for sending the sample to Norton.  This should help others experiencing the same aggravation.

73,
John WB2OSZ

Bigjohn4, 2025-08-27:
Thanks for the answers. I did get the file out of quarantine and was able to start it.
Now to do some reading and figure out how to use it  :-)

closed 2025-08-27 by Bigjohn4