wb2osz/direwolf #595: CON signal on Pi Zero 2W GPIO

w3wt, 2025-09-03:
I have an operational Digipi running Direwolf 1.8.  I added a fourth LED connected to GPIO-6 to indicate when the AX.25 TNC is in the Connected state.  According to the Direwolf user manual (p.72), adding the statement CON GPIOD gpiochip0 6 in the appropriate .conf file (here, direwolf.node.conf) is supposed to signal the status of the CON flag onto GPIO-6.  However, when connected to another station, Direwolf says I am connected but GPIO-6 is not asserted.  I can light the LED from the command line with gpioset 0 6=1.  So it appears that Direwolf is not working correctly.

dranch, 2025-09-04:
What version of Direwolf are you running and on what operating system version?  There have been major updates regarding gpiod in Direwolf in just the last few days to support the upcoming Trixie release:

   https://github.com/wb2osz/direwolf/commit/dce46d4446764de96a4823d13f2acfbca2c5fd87

Also, how are you keying up your radio?  Are you using a GPIO pin for PPT?  Are you using the DCD GPIO pin option as well?

w3wt, 2025-09-04:
Hi David. It is Direwolf 1.8. The OS is Bookworm. The DigiPi version is the latest, 1.9.4. For PTT, PTT GPIOD 12 is used in the .conf file to enable GPIO-12 to drive a MOSFET. That works fine. Similarly, GPIO-16 drives an LED for the DCD indicator and that works fine. Overall, the DigiPi works great. Thank you for all the work you do on Direwolf. 73 de W3WT

dce46d4

w3wt, 2025-09-04:
> Ok, your details look good.  Next, has this CON indicator ever worked for you and it later broke?  It's important to understand that this CON feature will only work with packet connections made via the AGW connector.  What packet client are you using to make these connected AX.25 sessions?  A KISS-based AX.25 setup won't work with this feature.

Hi David. The CON indication never worked for me but i have only been using the DigiPi for about a month. The LED/GPIO-6 hardware combination works though. When i enable the AX.25 node function and use the axcall utility (part of something else?) to connect to another station, the window that opens does say connected in the KISS mode so that may explain what i am experiencing. I don’t know what underlying packet client used by DigiPi is. But i copied Craig, km6lyw, on this email and he might be able to provide more information. 73 de W3WT

dranch, 2025-09-04:
The axcall tool is apart of the Linux AX.25 toolkit does NOT use the AGW connection to Direwolf, it's a KISS-based connection.    As such, this Direwolf CON feature will not work with this Linux command.  You can try using a different AX.25 connected mode application such as the following which all support the AGW protocol:

   https://github.com/mfncooper/paracon
   https://www.cantab.net/users/john.wiseman/Documents/QtTermTCP.html
   https://github.com/jmkristian/agwpe-tools
   http://uz7.ho.ua/packetradio.htm (via WINE)
   ipserial in https://www.outpostpm.org/index.php?content=downloads

w3wt, 2025-09-04:
Thanks, David. 73 de W3WT

https://github.com/mfncooper/paracon
https://www.cantab.net/users/john.wiseman/Documents/QtTermTCP.html
https://github.com/jmkristian/agwpe-tools